TauLabs/TauLabs #1442: OSX packaging messing up QtQuick again (with Qt 5.4)

peabody124, 2015-01-13:
As reported by balu, the welcome screen is broken again. There is a regression in macdeployqt for Qt 5.4:

https://bugreports.qt.io/browse/QTBUG-43100
https://bugreports.qt.io/browse/QTBUG-42952
https://bugreports.qt.io/browse/QTBUG-42954

http://forum.taulabs.org/viewtopic.php?f=18&t=480#p3846

peabody124, 2015-01-22:
For now there is a workaround for this. Compile Qt5.4.0 with the patch from here https://bugreports.qt.io/browse/QTBUG-42954 when packaging OSX. This is more appealing than continuing to hold OSX back a version compared to the other operating systems, and the fix should be applied in Qt 5.4.1.

peabody124, 2015-01-31:
I'm removing the blocker label on this because there is a workaround currently (recompiling Qt with the patch on macdeployqt). I'll leave it open until Qt 5.4.1 is released and we confirm that fixes the root issue.

mlyle, 2015-07-06:
Hack workaround merged in #1703; when we pick up 5.5.0 it should be reverted.

mlyle, 2015-08-29:
Fix merged, yaaay.

closed 2015-08-29 by mlyle