TauLabs/TauLabs #1445: GCS: Onlinehelp link to WIKI

cGiesen, 2015-01-19:
All links for online help in GCS to our wiki are broken.

Did someone change the wiki page names?

peabody124, 2015-01-19:
What are the current links?

cGiesen, 2015-01-19:
I test perhaps Modules, but the redirect is very fast.
So I have to search the code where I change all the URL last time....

peabody124, 2015-01-19:
It's worth noting that this "http://github.com/TauLabs/TauLabs/wiki/OnlineHelp:-Input-Configuration" seems to work, so the problem appears to be related to the wiki.taulabs.org -> http://github.com/TauLabs/TauLabs/wiki transformation

@solidgoldbomb do you know if this comes from the changes that github required on the DNS?

peabody124, 2015-01-20:
Yeah and Attitude has the real URL in the code: "https://github.com/TauLabs/TauLabs/wiki/OnlineHelp:-Attitude"

solidgoldbomb, 2015-01-20:
I'll take a look at this to see if I can fix the redirect.  The issue is definitely in the redirect:

```
$ http --verbose http://wiki.taulabs.org/OnlineHelp:-Attitude
GET /OnlineHelp:-Attitude HTTP/1.1
Accept: */*
Accept-Encoding: gzip, deflate, compress
Host: wiki.taulabs.org
User-Agent: HTTPie/0.8.0

HTTP/1.1 301 Moved Permanently
Connection: keep-alive
Content-Type: text/html; charset=utf-8
Date: Tue, 20 Jan 2015 17:55:08 GMT
Location: https://github.com/TauLabs/TauLabs/wiki
Server: nginx/1.6.2
Transfer-Encoding: chunked
X-Frame-Options: GOFORIT
```

The 301 redirect drops the trailing part of the URL as suspected.

solidgoldbomb, 2015-01-20:
Unfortunately, this appears to be a limitation of the dns host that I'm using for taulabs.org.  We can either change all of the URLs in the GCS or I can set up a more capable URL redirection service on my own server.

The quick fix might be to have a global URL base somewhere in the GCS code.  Then, add the tail of the URL to the base.  That way, you could fix it up by setting the URL base to point at github for now, then put it back to wiki.taulabs.org once the redirect is working better again.

peabody124, 2015-01-20:
Ok, I'll do that then for now. I might have to keep it hardcoded, though, because those paths are set in the designer files rather than programmatically.

cGiesen, 2015-01-24:
Thanks for fixing this error.

closed 2015-01-24 by cGiesen