TauLabs/TauLabs #1446: NEXT:   Firmware Upload Failed

thomaskausm, 2015-01-19:
Hello,

I updated my NEXT to the most current (commit 9256dd0b062c87056cca37e508aa21abd4ec1dc9)  and run the GCS and quanton make.
Then I updated the firmware and try to upload the quanton flight software via GCS. 

[image]

After pressing  "Flash" it tooks some seconds and the upload failed. 
[image]

The quanton board itself was touched/changed by that action in a way that all the LED's on the board shine (no blinking) and no change to connect the quanton by the GCS .

The only way to reanimate the board is to flash the flight software manual (e.g. openOCD and gdb)

I tried it serveral times and I saw the problem in 9 of 10 attempts so it seems to be a timing problem.
I searched the issues for "bootloader" problem and found https://github.com/TauLabs/TauLabs/pull/1385 and reverted that  change in my local build to the old value (500ms) but the firmware upload failed also.

Any idea how to get closer to the reason for that behaiviour ?

Regards
Thomas

fujin, 2015-01-20:
I was getting this and this may seem strange but switching the board to be directly plugged into my motherboard resolved the issue. If you follow `dmesg` while the flash is running, you may see something regarding USB resets, that is what I noticed and prompted me to switch ports.

thomaskausm, 2015-01-20:
Hi fujin,

thank you for the tip but using the usb ports of the mainboard directly  dosn't solve the problem. Also the logs offered by dmesg doesn't help. No USB resets/warnigs or something else.

This shows a regular connect 

[ 3379.139529] usb 2-2.1: new full-speed USB device number 63 using uhci_hcd
[ 3379.439692] cdc_acm 2-2.1:1.0: This device cannot do calls on its own. It is not a modem.
[ 3379.439717] cdc_acm 2-2.1:1.0: ttyACM0: USB ACM device
[ 3379.530256] input: Quantec Networks GmbH quanton fc as /devices/pci0000:00/0000:00:11.0/0000:02:00.0/usb2/2-2/2-2.1/2-2.1:1.2/input/input21
[ 3379.530620] generic-usb 0003:0FDA:0100.0036: input,hiddev0,hidraw1: USB HID v1.10 Joystick [Quantec Networks GmbH quanton fc] on usb-0000:02:00.0-2.1/input2

Now pressing halt 

[ 3379.530256] input: Quantec Networks GmbH quanton fc as /devices/pci0000:00/0000:00:11.0/0000:02:00.0/usb2/2-2/2-2.1/2-2.1:1.2/input/input21
[ 3379.530620] generic-usb 0003:0FDA:0100.0036: input,hiddev0,hidraw1: USB HID v1.10 Joystick [Quantec Networks GmbH quanton fc] on usb-0000:02:00.0-2.1/input2
[ 3482.204666] usb 2-2.1: USB disconnect, device number 63
[ 3483.162956] usb 2-2.1: new full-speed USB device number 64 using uhci_hcd
[ 3483.475398] generic-usb 0003:0FDA:0100.0037: hiddev0,hidraw1: USB HID v1.10 Device [Quantec Networks GmbH quanton fc] on usb-0000:02:00.0-2.1/input0

GCS shows now "Connection to bootloader successful"
Now loading the new firmware and press "Flash"

Firmware upload failed !!!  

The strange think here is that no new log entries are show by demsg, Entry [ 3483.475398] is still the last.

I also had the situation that flashing the firmware works three times in a row but the forth time and every further try failed. I never saw this kind of problem in the past. I run the tests within a VM-Ware image (as always). Running the current MASTER  GCS works fine.

Any other ideas which leads to the cause of the problem ?

Regards
Thomas

thomaskausm, 2015-01-21:
Hello again,

it seems to be a VM-Ware/virtualization issue. On the same host I run my VM-Ware image  I now built the current NEXT.  There was no problem to flash/boot ... the firmnware form the host itself but  I run directly into the "upload failed"  problem as I switched to the VM-Ware image running on the same host (in both cases I used the sam USB port, same cable :-)).

For whatever reason that behaiviour happens only with the NEXT version but not with the MASTER.
But anyway the problem is (more or less) clear to me and I will close that report.

Sorry for the "noise" I have made

Regards
Thomas

closed 2015-01-21 by thomaskausm