TauLabs/TauLabs #1449: Feature: Helicopter Flybar Mix

mmartinmate, 2015-01-22:
Hello,

First of all congratulations for the very good job you have done.

I would ask if you are planned to include mixes for traditional helicopters with flybar?

I have seen that APM includes this feature.

I don't know if that is very complex or could be done easily.

Yours sincerely,
Miguel.

peabody124, 2015-01-31:
In the short run, I don't think it is too likely. It isn't a question of calculating the appropriate mixer. Rather, because of the slow response due to the mechanical flybar our current control code does not work very well.

peabody124, 2015-01-31:
P.S. I closed this because it was a question. If you want to rephrase your question as a feature request we can reopen it.

closed 2015-01-31 by peabody124