TauLabs/TauLabs #1468: GCS: temperature calibration stack overflow crash after 30 minutes

taumcg, 2015-01-30:
20150124 build on Win7-64.

Connected chilled Sparky to GCS and ran temp calibration for 30C range.  When it reached 100% after about 30-40 minutes, before I could click Accept, GCS crashed with stack overflow:

Problem signature:
 Problem Event Name: APPCRASH
 Application Name: taulabsgcs.exe
 Application Version: 0.0.0.0
 Application Timestamp: 02efcfb8
 Fault Module Name: Qt5Core.dll
 Fault Module Version: 5.4.0.0
 Fault Module Timestamp: 777a675f
 Exception Code: c00000fd
 Exception Offset: 00244664
 OS Version: 6.1.7601.2.1.0.256.48
 Locale ID: 1033
 Additional Information 1: bbff
 Additional Information 2: bbff281b59e6a9c73a384aeb40c8ad5f
 Additional Information 3: d1f1
 Additional Information 4: d1f16d62a3ddecd69f1b29f2eca4a2ee

peabody124, 2015-01-30:
I'm guessing from your backtrace this was on windows. What version of the software? (githash is most useful)

taumcg, 2015-01-30:
Yes, Win7-64.

 

20151024 from jenkins dist, GIT Hash 03bcc8cb.