TauLabs/TauLabs #1469: GCS Linux 64Bit, wrong tooltip background color

LinuxJohannes, 2015-01-30:
The background is darkgray, the characters are black. Very hard to read the displayed text.

LinuxJohannes, 2015-10-25:
Found a solution. There are some HTML-Files to display the message. For example, Batery-Error:

Battery-Error.html
```<html>
  <head>
    <title></title>
    <meta content="">
    <style></style>
  </head>
  <body style="color: white">
    <h1>Battery: Error</h1>
    <p>
    Both battery current and voltage are at or below zero.
    </p>
  </body>
</html>

```
You can find the files in ground/gcs/src/plugins/systemhealt/html/
Changed the color from black to white, problem solved on my system. Can someone verify it on windows and/or on Mac please.

kubark42, 2015-10-26:
@LinuxJohannes Could you post before and after screenshots? It would be good to check that the solution looks the same across platforms.

LinuxJohannes, 2015-10-26:
http://forum.taulabs.org/download/file.php?id=385
Thats a picture from bevore. Text is now white on a dark gray background

kubark42, 2015-10-26:
Thanks. Is this all tooltips, or just the alarm tooltips?

The alarm tooltips look like this on OSX:
[image]

mlyle, 2015-10-26:
White on yellow would be bad.  Maybe if we jsut don't specify the color the right thing will happen?

tracernz, 2015-10-26:
Uhh, the alarms appear correct (the same as @kubark42's photo) on my Ubuntu 14.04 64-bit machine?
[image]

mlyle, 2015-10-26:
I don't remember the combo, but I've seen the black-on-black tooltips before.

tracernz, 2015-10-26:
Ok, I think a PR that explicitly sets the colours correctly in the CSS won't cause any harm so let's do it. :)

LinuxJohannes, 2015-10-27:
That's only on the alarm tooltips. Haven't seen it on another tooltips. Better to set the background collor?  
By the way. The translator can't translate this tooltips. Better is to include the tips in to the source code.

PTDreamer, 2015-10-27:
> By the way. The translator can't translate this tooltips. Better is to include the tips in to the source code.

It would be easier just to include the translated html files in a subdirectory and handle that on the gcs code.