TauLabs/TauLabs #1472: Missing DLLs in Master Windows from 2015.01.31

cGiesen, 2015-02-01:
taulabs_master_20150131_232019_4082e9d88c_win32

After starting GCS on Windows 8.1 x64
[image]

peabody124, 2015-02-02:
Installer or portable version?

cGiesen, 2015-02-02:
I have only extract the file I download here:
http://jenkins.taulabs.org/artifacts/taulabs_master_20150131_232019_4082e9d88c_win32.exe

guilhermito, 2015-02-04:
Are you sure the plugins are missing? That message looks like it is trying to load plugins from a previous build.

mlyle, 2015-06-18:
I ran this build successfully on windows 8.1 x64.  Does this still happen, @cGiesen, or were you able to make it work??

mlyle, 2015-07-03:
Closed INVALID due to lack of details.

closed 2015-07-03 by mlyle