TauLabs/TauLabs #1473: Error enabling Autotune on Quanton, Sparky, CC3D and CC

pug398, 2015-02-02:
Pre release 20150131-4082e9d8 Win7 32bit

Opening this to get it out of the way. Quanton and Sparky give System Ident error on autotune enable but module does enable. Warning is given on first save attempt and red X on second.

Autotune enable has generated Boot Fault on CC3D and CC since release after 10_26_14. Works on 10_26_14 does not work on 11_26_14 if it helps.

peabody124, 2015-02-02:
Thanks for the report. What do you mean they give "System Ident error on autotune enable?

peabody124, 2015-02-02:
Ok I see it from a screenshot that was attached on forums (duplicated here).

[image]

That is a purely UI bug. The autotune page does not need to save or even send the SystemIdent settings. Previously it was a data object so the widget did not even try, but we changed to a setting so people could power down after running SystemIdent.

@guilhermito do you have any suggestions? I'd like to keep the automatic signals from the smart system but not have it try and apply or save the ```SystemIdent``` object. If there isn't an easy workaround I can just directly update the fields based on the object...

peabody124, 2015-02-02:
I created a separate issue for CC3D autotune.

guilhermito, 2015-02-02:
Sorry no time to look at this right now but it should be fixable just by using "setNotMandatory" created on #1386

pug398, 2015-02-02:
I see you are already on top of it. Thanks.

mlyle, 2015-09-10:
Fixed by #1650.

closed 2015-09-10 by mlyle