TauLabs/TauLabs #1476: Enabling AutoTune on CC3D results in a boot fault

peabody124, 2015-02-02:
As reported in #1473 (but separated out into its own issue here), enabling Autotune with CC3D results in a boot fault. I'm guessing with MWRate mode included CC3D is running out of memory. This might be a persistent problem until Chibios, but perhaps we can shuffle things around.

CC3D is really starting to shows its limits ...

pug398, 2015-02-03:
I will test pre release over next 24 hours. To clarify I did not run out of memory with CC3D due to lack of memory. When firmware is put on board and booted there is 164 bytes for IRQ stack, 2084 for heap. It will at intervals push something onto the stack and not pop it off so over time it eats up the IRQ stack memory but heap is still at 2084. Looks like when IRQ stack falls below 100 bytes it gives memory warning. That was just before pre release so will let that run on CC3D tonight to see if it still does it.

Looks like it just dropped 40 bytes of stack while typing that in a few minutes.

awlx, 2015-02-20:
Any news on this issue? :)

ernieift, 2015-02-26:
fixed in #1495.

mlyle, 2015-06-18:
Can anyone confirm that CC3D is indeed able to autotune now?  If so we'll close.

pug398, 2015-06-18:
Last I checked autotune was broken on CC. Boot fault. Not sure about CC3D but I will try to check both tonight.

mlyle, 2015-06-18:
Thanks, I appreciate it :)  Trying to clear out issues so we have a useful list.

ernieift, 2015-06-18:
Same here. I tried to get some ram from other modules stack. No luck yet.

mlyle, 2015-06-18:
We could try enabling -fconserve-stack on some modules or some targets...

pug398, 2015-06-18:
Just checked and both CC and CC3D get boot fault. Hopefully memory can be freed up somewhere.

Trex4Git, 2015-06-18:
Maybe it is worth to optimize the CC3D towards Acro / FPV racing stuff. I don know, but maybe then some modules can be disbaled to free some memory.

pug398, 2015-06-18:
I agree. If it is an unresolvable memory issue I would scrap advanced flight support on them and make autotune functional. My CC boards run 70-80% cpu usage just sitting on the bench with no extra modules running. This keeps coming up and the devs have done their best to keep it going this long.

pug398, 2015-06-18:
Perhaps this is not directly a memory issue. Someone else will have to check behind me but it appears TL3_23_15 can enable autotune and boot but TL3_24_15 causes boot failure after autotune module enabled. Believe that was when we moved to ChibiOS.

mlyle, 2015-07-03:
This is addressed by PR #1680 once it lands.

mlyle, 2015-07-06:
#1680 merged; once there's a little bit of positive test feedback this can close.

mlyle, 2015-07-07:
Positive feedback! Closin'.

closed 2015-07-07 by mlyle

Oliv4945, 2015-08-22:
Hi. I just pushed Taulabs to a CC3D on a tricopter.
All default config, CPPM, I tried with and without telemetry on main port (bluetooth)

The heap remaining is 1368 bytes, so when I enable the autotune I have the same boot fault. I am using GCS 20150713: 20150713-4b9fc7dc and firmware tag 20150713 , git tag 4b9fc7dc

Should I try to build it myself rather than using your release ?

mlyle, 2015-08-22:
@Oliv4945 I don't have a CC3D in front of me so these numbers are from memory.

Enabling autotune uses ~1684 bytes of heap; so you'd normally want to have 1744 or so free beforehand.  20150713 is indeed the release and people had success flying it with PPM.

Please confirm you have nothing else using a serial port (telemetry buffers will make this impossible) and no modules running except Autotune.  After completing the autotune flight, you can turn the autotune module off and other modules back on.

The other potential problem is if you have optional configuration objects from an old configuration-- they can use up ram.  If you can't get to >1700 bytes of free heap by turning off modules and ports, you may have to erase flash.

Oliv4945, 2015-08-22:
@mlyle Sorry for the bad information. You are right, there is enough memory to activate it.
I probably forgot to reboot the CC3D when I tried to deactivate the telemetry link

gregorydulin, 2016-01-05:
Thanks for all the commentary, it helped me out a lot.  I just a CC3D Atom to boot with AutoTune and PPM enabled.  Here's what I did:

Flashed TauLabs fw_coptercontrol.tlfw (20151123 76f58893)
Erased all settings
Set PPM input (saved)
Ensured serial ports were disabled (saved)
Set 2 flight modes, one of them being AutoTune (saved)
Enabled AutoTune module (saved)
Disconnect
Rebooted by unplugging and plugging in USB cable.
Booted!

Memory was in "Warn" state, but it boots.  I set up vehicle type, inputs, outputs, calibrated ESCs and flew it for a minute or two (it's cold and dark outside).  I'll give AutoTune a shot tomorrow when the sun comes back.  I'm really looking forward to it.

Thanks!

mlyle, 2016-01-05:
Autotune doesn't work on Tau on F1 targets because of lost points and jitter.  The event system stuff htat was merged in dRonin is necessary.

gregorydulin, 2016-01-06:
@mlyle Sorry, I'm not up to speed on the Flight Controller vocabulary enough to really know what you mean by "F1 targets", or "lost points" (my fault, not yours).  I went through the AutoTune process and ended up with PID settings that did seem a little "jittery".  I was able to tame it a bit with the Damping and Noise sensitivity sliders, but it's still a bit twitchy.  Is that what you mean by "doesn't work"?

Tomorrow, when I get another 15 minutes in the sun, I'll compare the (damped) AutoTune settings to the Default settings.  If AutoTune gets me closer to "good" settings than Default, I'll still consider it a win.

mlyle, 2016-01-06:
@gregorydulin CC3D and Naze32 have a STM32F1 microcontroller which doesn't have much RAM or a floating point unit.  When I left Tau I was doing work to make the autotune process more efficient so it could fit adequately on these targets-- including the issue closed here that freed up a bunch of RAM.

But it wasn't enough.  The autotune system ends up getting bad information (lots of missing observations of what's happening and duplicated points).   The PIDs that it produces are not good and not anywhere near a real measurement of the system like autotune is supposed to produce.

I'm now doing work over at dRonin (a Tau fork focused on quicker advancement where we've gotten autotune working on F1).  A prerelease is at https://github.com/d-ronin/dRonin/releases/tag/preview-20151218 and we'll be shipping our first release in a week or two.

raemin, 2016-01-13:
I can confirm what mlyle stated: I am very new to quads and just bought a noname quads (eachine 250 racer actually) with "so so" setup, so decided to upgrade from openpilot to taulabs, to discover that the board was consistently disconnecting from the GCS when using a PWM protocol, and would not connect at all with Autotune enabled. Based on mlyle message I just tried the above mentioned release. With PWM and autotune I can only have 5 PWM channels (6 channels makes the board unsable), but autotune does work great! 

Too bad the settings from Taulabs are not interchangeable with the ones from dRonin