TauLabs/TauLabs #1484: Quanton board not recognized on OSX

crazymao, 2015-02-07:
I followed the instructions here: https://github.com/TauLabs/TauLabs/wiki/Development-OSX-setup it worked without fail, everything downloaded and compiled correctly. However, neither the GCS or the CLI tools are recognizing my quanton board, I get the following message whenever I try to execute a command `make [bl,fw,ef]_quanton_program`:
```
Open On-Chip Debugger 0.8.0 (2015-02-06-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.sourceforge.net/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 1000 kHz
Error: open failed
in procedure 'transport'
in procedure 'init'
make[1]: *** [program] Error 1
make: *** [ef_quanton_program] Error 2
```
Strange since it does get recognized on my Windows VM so at least I know the board is okay. Any ideas?

ulab, 2015-02-08:
My Quanton is recognized in the GCS on OS X 10.10.2. But I'm not using the development setup, just the basic GCS.

Is you board recognized at all? A few seconds after connecting I see the following two lines in /var/log/system.log:
```
Feb  8 12:15:52 localhost kernel[0]: AppleUSBCDCACMData: Version number - 4.2.2b5, Input buffers 8, Output buffers 16
Feb  8 12:15:52 localhost kernel[0]: AppleUSBCDC: Version number - 4.2.2b5
```

crazymao, 2015-02-08:
I can see those exact lines as well, but the GCS doesn't recognize it. It just says `Connections: Serial: usbmodem1411` instead of `quanton`, I get no telemetry and get a `No Link` error/warning.

peabody124, 2015-02-08:
I just want to make sure I understand. Those commands ```make bl_quanton_program``` are to flash the code via st-link. Not to do anything via the USB port. Was this a previously working board that you are trying to reflash? Was it ever working?

You probably should post some details about your wiring of the programmer since that is probably where the problem lies. You are using st-link?

crazymao, 2015-02-08:
After doing further research I realized that those commands don't work over USB, my bad I apologize I'm pretty new to this still trying to figure things out heh. The board works just fine, I will be writing custom code for it however so I was trying to figure out how to flash code to it. Is it possible via the USB link without any extra peripherals? How does the GCS upload commands/configurations to the board? Thank you

guilhermito, 2015-02-08:
This issue was already discussed on the forum. This is just a user error, closing.

closed 2015-02-08 by guilhermito