TauLabs/TauLabs #1496: Remove "Max rate attitude (deg/s)" from GCS GUI

peabody124, 2015-02-28:
This should probably stay as a parameter available in the browser, but
there is never a real time that the end user needs to adjust this to get
a desired flight style. In fact, I always have to look up what it does by
going through both the GCS code and the flight code.

Basically we want to keep the rate mode max rate and max angles.

elafargue, 2015-03-26:
... by the way: what _does_ it do ? :)

peabody124, 2015-03-26:
in attitude mode it limits the rate that the outer loop can request. It isn't very common to hit 300 deg/s in attitude mode though so it has little practical effect.

elafargue, 2015-03-30:
OK, will update GCS UI, should be pretty straightforward.

elafargue, 2015-05-06:
This is now fixed.

closed 2015-05-06 by elafargue