TauLabs/TauLabs #1507: Logging module: file header invalid

BrainFPV, 2015-03-09:
I have been trying to use the new logging module on the Brain FPV FC, but when I download the log and try to open it using the Python parser, I get an error stating the header of the file is invalid. Has anyone else experienced this problem or are extra steps needed to open the file?

BrainFPV, 2015-03-10:
ping @peabody124 when logging with Android, I see a header of the form below, but it isn't present when logging with the internal module. Is this expected? 

    Tau Labs git hash:
    66d34c53
    95ed7f467993fa474dda95392d0c77e6b59efd4e

BrainFPV, 2015-03-22:
The log file is valid, but the header is missing. I can read it using a modified version of the Python parser.

A solution would be to write the version info at the start of the internal log file and the GCS formats it properly when exporting. Thoughts?

peabody124, 2015-03-22:
@BrainFPV yeah that sounds like a great idea!

BrainFPV, 2015-03-23:
I will look into it. 

Also, I had some ideas how to make logging more useful for regular users. First, it would be nice if the GCS would show a list of logs and the date when the flight occured. To get the date, we could add some functionality to the logfs to get the GPS time for each log file (it would just look for the first good GPS time entry). Second, the GCS should offer the option to export the log as a KML file (it would just parse the GPSPosition objects and export the results to the KML format).

peabody124, 2015-03-23:
Yeah I think both of those would be quite useful. Another option would just be a python script that essentially does that to a bunch of log files in a directory. But integrating into GCS is probably the most user friendly thing.

elafargue, 2015-03-24:
@BrainFPV I love the KML export idea - what I would like to see is an option to include additional UAVObjects inthe KML export: for instance, if you want a KML that also contains your battery level, or barometric pressure or other. Would be super useful.

elafargue, 2015-03-24:
@BrainFPV there is a KML export plugin in the GCS that is disabled by default, have you tried it? IMHO if this is simply a log file conversion plugin, it should be either moved as a separate utility outside of the GCS, or it should be turned into another type of log export format ? Thoughts? Shall we open a separate issue on this?

PR #1682 (2015-06-27): Logging module improvements
Adds the following improvements to the Logging module:

- Writes correct TauLabs log file header. Fixes #1507
- Adds the option to log all settings UAVOs at the start of the log file. This information will be very useful when analyzing logs. Also, the space needed for the settings is small, so it is enabled by default.
- Logs some UAVOs on change (currently only `FlightStatus` and `WaypointActive`)
- Adds some additional UAVOs to be logged
- Fixes a bug in `pios_streamfs` that limited the log file size to 1MB (`uint8_t` was used for `arena_id`)
closed 2015-06-27 by peabody124