TauLabs/TauLabs #1509: GCS: Check for enabled modules depending on selected flight modes

BrainFPV, 2015-03-10:
At the moment it is possible to select ,e.g., the PositionHold flight mode without having the path follower module enabled, which results in a configuration error. For new users, it is not very obvious which modules have to be enabled for a certain flight mode.

I think it would make sense to add some logic to the flight mode selection that tells the user which modules must be enabled if they are not already enabled. Thoughts?

BrainFPV, 2015-03-22:
Thoughts?

pieri70, 2015-03-23:
For me nice idea, I did't knew why my quanton did't arm after PH added to FModes..

ulab, 2015-03-23:
I thought about something like this too when I fiddled around with the modules. Some kind of dependency mechanism like the Linux distributions have. Each module could have some kind of "Required: X, Y, Z" and "Conflicts: A, B" to make that possible.

peabody124, 2015-03-23:
I'm not sure anything is needed in the firmware, this logic can probably be handled in GCS. Possibly within the system health gadget. We already throw the SystemConfig error and there is the "ConfigError" field of SystemAlarms which basically says the module that is missing. That information just isn't easily conveyed to the user.