TauLabs/TauLabs #1515: Sparky after Chibios: enable GPS and bootloop

sfabris, 2015-03-24:
So,

I've able to replicate this:
- clean sparky (erase all settings from board);
- flash next (after chibios merge)
- reboot
- enable GPS module and set an uart to GPS (i.e. Flexi)
- reboot
- bootloop.

If I try rescue, then safe boot: also bootloop.

I'm sorry but I don't have the knowledge to help on this more than opening this issue..

peabody124, 2015-03-24:
Thanks for the report! I'll look at it soon

peabody124, 2015-03-25:
Ok. This is because ChibiOS doesn't play nicely with stack sizes that aren't ```aligned(8)```.  Working on a fix that will prevent this error again.

peabody124, 2015-03-26:
@sfabris can you confirm this is fixed on next for you now?

sfabris, 2015-03-26:
Works properly!. 
Thanks for the fix.

closed 2015-03-26 by sfabris