TauLabs/TauLabs #1517: "FlightTime modul critical" prevents arming for 60s~80s with voltage sensor only

gompf, 2015-03-25:
Sparky1, if booted wo battery monitoring module red status led goes of with the "third beep from blheli", comes again for about 2s and then everything is fine. 
With Battery monitoring and only voltage sensing configures red led stays on and "FlightTime module critical" prevents arming. If I just let it sit there for 60s-80s red led goes off/module warning dissapears and I can arm&takeoff. 
I havent applied the stack boundary patch yet.
Modules Battery Monitoring and Altitude Hold; oneshot on 1,2,4; mpu666@53%cpu; 5368 bytes heap remaining

joergrohde, 2015-03-25:
Sparky1 taulabs_chibios_20150321_170854_290690f58f
With Battery Voltage only arming is possible after 70s
Modules Bat, Hott, OneShot125@1-4, mpu@666

elafargue, 2015-03-26:
Can you report stack remaining on the battery module? I noticed it can be super low (4 bytes only in my case), might be related?

gompf, 2015-03-26:
Within 3 boots I have seen 56,40 and 8byte (TaskInfo/MetaData/Stack remaining/Battery). It doesn't change when it's ready to arm. Arming with 8byte left was OK, after 70s. 
I found chibios branch now (didn't noticed the switch to next) and will test again with current version later.

joergrohde, 2015-03-26:
Stacksize 40, 16, 48Byte. Current pin is set to none, Current measurement is off. But live reading -1A and a Capacity left 146xxxxxxxxxxxxxxxxx mAH.

gompf, 2015-03-26:
Forgot to mention it, same config. I also assume a rollover or something in cap. calculation when current measurement isn't configured.

elafargue, 2015-03-30:
@joergrohde @gompf does the latest fix by peabody124 resolve your problem ?

gompf, 2015-03-30:
Yes, everything fine here. Tested in modul combination BatteryMonitoring/AltitudeHold/FrSkyHub. Thanks!

closed 2015-03-30 by peabody124

joergrohde, 2015-03-30:
Yes, Sorry only desk tested. Modul Batt, Hott on Quad & Batt, Hott, Autotune on Hexa. The weather here where I´m living in Germany is bad :(