TauLabs/TauLabs #1524: Make PWM/OneShot selection configurable in GCS

BrainFPV, 2015-03-28:
We should have an user friendly way to enable OneShot in the GCS. One option could be to add a selection to the output channel config as shown below. Thoughts?

[image]

joergrohde, 2015-03-29:
That would be nice. But also the true value. When you select OneShot the factor 12 has to be calculated in the GCS in the right way. So one has to make his settings 1000-2000 or 125-250, imho.

peabody124, 2015-03-29:
Agreed to both these points. More specifically, I think the settings should apply at the top the way update rate does since it influences an entire timer bank. This will avoid confusing situations where two channels on the same timer are assigned different tasks which will produce unexpected outcomes.

joergrohde, 2015-03-29:
You meen at this place?
[image]

BrainFPV, 2015-03-29:
I will look into adding it to the update rate selection. This makes me wonder why OneShot was added as an option to `ChannelType` and not `ChannelUpdateFreq`. As @peabody124 said, configuring some channels of a group as OneShot and others as PWM will produce unexpected outcomes.

How about changing `ChannelUpdateFreq` to be an enum and list OneShot there? This means it would be removed from `ChannelType`.

peabody124, 2015-03-29:
See the discussion in #1424 for rational. TL;DR: step 1 get it in and tested, step 2 refactor for user friendliness. I've been meaning to work on it (and actually consider this a release blocker, or at least that we don't announce OneShot).

@ernieieft might want to comment here.

The enum option is a bit difficult - for example of the brushed sparky I run at 8000 Hz output, so the freedom to freetype is nice. I thought a value of 0 or the max value might be appropriate for indicating OneShot. The GCS already maps that field to an enum combo box so from the GCS perspective it can end up in the same combo box.

It would also be nice to have a way of indicating x12 resolution while not using OneShot. For example again on BrushedSparky I'd prefer to run to even higher update rates with higher resolution, but not synchronously updating. In this case, overloading the Update Rate field might be different than the timer mode. We could have three timer modes:

 - x1 resolution, periodic
 - x12 resolution, periodic
 - x12 resolution, OneShot

Also I agree with @joergrohde that showing real time values in the GCS would be nice. My preference would be to even use them consistently in the ActuatorSettings and keep it always in µs, but have the Actuator module apply the scaling to the settings there. This will be easier when the whole bank is in the same mode.

BrainFPV, 2015-03-29:
@peabody124 makes sense. I like the idea of using `UpdateFreq=0` for OneShot. Regarding the timer resolution, would there be downside to always using 12x resolution? I agree that it would make sense to have the values in `ActuatorSettings` always use `us` as unit (this doesn't mean the timers will run at a lower resolution, just min/max/neutral can only be configured with `us` resolution).

How about this:

1. We use `UpdateFreq=0` to indicate `OneShot` and remove it from the `ChannelType`. This is added as an option to the GCS to make it easy for users to use OneShot.
2. We add `HRPWM` (high-resolution PWM) to `ChannelType`, which would indicate normal perodic PWM with 12x resolution. This option is not exposed to users (only through object browser). This would not be necessary if we always use PWM with 12x resolution.

I will take a stab at this, but let's first agree on the best way to implement it.

BrainFPV, 2015-03-30:
ping @peabody124 @ernieift

peabody124, 2015-03-30:
So there are two consequences of always going to 12 Mhz for the timers:
 - The PPM / PWM capture requires the overflow handler to keep track of the higher bits of hte timer. Currently those run at ~15 Hz and this would make them be called at 188 Hz. This burns a bit more CPU. That being said, the timers _should_ handle this gracefully anyway so we don't have the issues that setting an output to HRPWM kills the inputs.
 - The output timers use a uint16 for the timer pulse duration. At 12 MHz that means the maximum pulse duration is 5 ms. I'm not sure this has any negative practical implications.

TL;DR: i think we are ok to put the output only timers at 12 MHz. For the input or shared timers it is a bit more questionable.

joergrohde, 2015-03-30:
That meens on CC3D, Sparky, Revo 6 Outputs and on Quanton 8, right?
Do one need OneShot125 on more than 6/8 outputs?

BrainFPV, 2015-03-30:
OK, I will make the high-resolution PWM optional, so we can test it and if it works well make it the default at some point.

ernieift, 2015-04-01:
Sorry for the delayed answer. Had to work.

I wouldn't change the update frequency to enum. You can still use a PWM out as signal generator this way.
If you want to use hires PWM with a fixed rate you have to make sure that the 16bit timer isn't overrun at low frequencies. This is more a problem on F1.
Using a normal ESC in sync to MPU is possible, if the MPU rate is <400Hz. You can use 12000..24000 as min/max for this. If you implement a way to skip a impulse when a previos one is still running, you can set the MPU rate higher. But this thwarts the advantage we have from OneShot. You may test older ESCs in normal PWM mode with OneShot enabled at 333Hz MPU rate. I guess they perform better as on 400Hz in asynchronous mode.
Sadly it is difficult to prevent a mixed config of PWM and OneShot at the same Timer. If it were easy, i had it implemented already. There are many checks which can overload the firmware without a real benefit. On the other hand you are able to use a more special config. This could be necessary for a few vehicles.
I think the best way to help users is to make OneShot access easier in the gcs.
The idea on the first picture here from @BrainFPV is very good. The gcs could show MIN/NEUTRAL/MAX divided by 12 if OneShot is selected. So we have all possible output types (also i2c etc.) in the output page. All what the gcs has to check is the channels against the output types when the user clicks apply/save. A simple warning for a faulty config would be enough. Not all configs may be wrong here.
I haven't looked closer but somewhere in the gcs (quanton.cpp etc.) the channelBanks are described. There are no UAVOs for this. So it is easier on gcs side to check possible timer issues.

Setting Freq=0 for the general conversion to OneShot brings a problem in I2C ESCs. Sure, we can always use the 1000..2000 range and multiply with 1.5 in OneShot but this raises CPU usage and cuts the resolution.

peabody124, 2015-04-07:
I would point out, we don't need resolution for the configuration. E.g. we can specify the range as 125 to 250 µs in real units, but inside the firmware it can use the x12 resolution. I don't think anyone needs sub microsecond resolution for the end points.

Also if you look at how the dropdown for channel update rate currently is, it already fakes an enum even though on the backend we are storing a real rate. We can do the same thing - when using OneShot (specifically the synchronous aspect) the dropdown can say that. In terms of the UAVO it can be setting the ChannelUpdateFreq 0 and a different field like ChannelResolutionType (which is grouped by timers like ChannelUpdateFreq) to HWPWM.

@BrainFPV have you made any progress on this?

peabody124, 2015-04-17:
@BrainFPV any update here? I can take a crack this weekend if you haven't had the time.

ernieift, 2015-04-17:
If I find some time over the weekend, I would try to change ActuatorCommand so that float is used instead of uint16_t. It could be used to keep the microsecond values and have a hardware-independent way to increase the resolution.

peabody124, 2015-04-17:
So internally I think we should be computing the mixer with floats to account for the fact we will be using sub-microsecond. I'm not sure about the value of changing ActuatorCommand - that is mostly a debugging object relayed to the ground side (it is not consumed by another module).

The main impact of changing ActuatorCommand would be doubling the data size that is transferred when we update the outputs which just burns bandwidth... Do you think for debugging we need that kind of resolution (plus we also have MixerStatus).

ernieift, 2015-04-18:
The set_channel() function is fed with the values from ActuatorCommand (Set/Get). We could change the return value of scale_Channel() to float type thats all. There is already the status[i] as float.
I know ActuatorCommand seems to be only for monitoring. To get the higher resolution we have to change the ActuatorCommand behavior or find a workaround.

ernieift, 2015-05-02:
@BrainFPV thanks for the picture above. I tried the look in #1566 but the idea is yours.

peabody124, 2015-05-18:
will be fixed by #1615

solidgoldbomb, 2015-05-26:
Closing this.  Should be fixed by #1615.  Feel free to re-open if it's not actually fixed.

closed 2015-05-26 by solidgoldbomb