TauLabs/TauLabs #1533: Import settings messed up ManualControlSettings for reversed channels

peabody124, 2015-03-31:
I've noticed when importing settings that have inputs that are inverted, the import fails the first time. I think that is because of a race condition on settings min, neutral, max in that order. One solution would be a hack in the import wizard that applies that setting twice.

A better solution would be if the input slider gadgets (which I'm pretty sure cause the bug) were robust to an atomic update that produces valid samples.

lolorc, 2015-04-05:
just reproduced it, export settings, erase all settings from board, import

            <field name="ChannelMin" values="172,172,1811,172,172,0,0,0,0,0"/>
            <field name="ChannelNeutral" values="204,995,988,986,991,0,0,0,0,0"/>
            <field name="ChannelMax" values="1811,1811,172,1811,1811,0,0,0,0,0"/>

=> Pitch got its neutral set to  1811

peabody124, 2015-05-18:
Fixed by #1574

closed 2015-05-18 by peabody124