TauLabs/TauLabs #1536: F4 targets fail to build on windows OS after chibios merge

jbtecno, 2015-04-01:
It may be environment specific but since the merge I get a "cannot open map file" when I try to build F4 targets.
Curious that one of google hits regarding that error is chibios related https://groups.google.com/forum/#!topic/swiftnav-discuss/P4A_WVpP810

One easy fix is disable the map file creation on the targets Makefile's

elafargue, 2015-04-04:
Can you provide more info on your OS ? Please delete the "build" directory entirely, make sure the arm sdk is at the latest version (make arm_sdk_install), and retry...

jbtecno, 2015-04-04:
Windows7 64bit, using msygit 1.9.5, python 2.7 and arm 4.9_2015_q1.
Build directory clean and fresh clone give the same results. This is a VM running on an esxi server and is a node of a jenkins CI system.
I have tried a clean windows7 install without any updates and with only the required tools and the results are the same.

peabody124, 2015-04-06:
Can you post a build log?

jbtecno, 2015-04-06:
Windows clipboard seems to be to short for the full log. Here is part of the output of "make quanton V=1"
http://pastebin.com/vx0vxcjM

Also that is after fixing some whitespace issue which results in this:
http://pastebin.com/iz3Q7qEM

jbtecno, 2015-04-06:
Looks like I'm not alone http://forum.taulabs.org/viewtopic.php?f=17&t=629&start=20#p5266

peabody124, 2015-04-06:
Can you try moving the directory to /c/ - you might be running into path length issues

jbtecno, 2015-04-06:
I have, same error.

jihlein, 2015-04-14:
Here's my build paths and tool versions:

call :which QTMINGW         "C:\Qt\Tools\mingw482_32\bin"              mingw32-make.exe
call :which QTSDK           "C:\Qt\5.3\mingw482_32\bin"                qmake.exe
call :which ARMTOOLCHAIN    "C:\GNU Tools ARM Embedded\4.8 2013q4\bin" arm-none-eabi-gcc.exe
call :which PYTHON          "C:\Python27"                              python.exe
call :which UNSIS           "C:\Program Files (x86)\NSIS\Unicode"      makensis.exe
call :which OPENOCDBIN      "C:\OpenOCD\0.4.0\bin"                     openocd.exe

GIT version is 1.9.5-preview20150319.  Git is located in C:\git.  Code is located in C:\Taulabs.  Operating system is Windows 7 Professional 64 bit, CPU is Core i7-4710HQ, 2.5 GHz, with 12 GB RAM.

Can not build F4 targets after chibios merge.  All files compile, but link fails with segmentation fault as documented in chibios merged thread on forum.

F3 targets build correctly. I was even able to port chibios to the Naze32Pro target following Sparky as a guide.  It also builds and runs fine.

Let me know if there's anymore info needed to troubleshoot this problem.

jihlein, 2015-04-16:
I have successfully built F4 target with ChiBios now!  Using GNU Tools ARM Embedded 4.9 2015q1 and Git-1.9.5-preview20150319, along with the following modification to the Quanton and Colibri makefile.  The line:

CDEFS += -DCORTEX_VTOR_INIT="($(FW_BANK_BASE) - $(EF_BANK_BASE))"

needs to be changed to:

CDEFS += -DCORTEX_VTOR_INIT=0x00020000

and then all the errors are cleaned up.  I know there has got to be a better way to do this, but I'm not well versed in makefile syntax.

This worked on Win7 Professional 64 bit, not sure if it would break Linix or OSX  building.

jihlein, 2015-04-17:
The better way is changing the double quotes to single quotes:

CDEFS += -DCORTEX_VTOR_INIT="($(FW_BANK_BASE) - $(EF_BANK_BASE))"

to:

CDEFS += -DCORTEX_VTOR_INIT='($(FW_BANK_BASE) - $(EF_BANK_BASE))'

Can anybody confirm this will work in Linux and OSX environments?

peabody124, 2015-04-17:
I can confirm it compiles correctly on OSX. However, the binary tlfw file is not identical so that change is having an impact...

peabody124, 2015-04-17:
ok but putting this into chcore_v7m.c:

```
#define STRINGIFY(s) XSTRINGIFY(s)
#define XSTRINGIFY(s) #s
#pragma message ("CORTEX_VTOR_INIT" STRINGIFY(CORTEX_VTOR_INIT))
```

it appears to have the same value with both approaches.

peabody124, 2015-04-17:
Would you mind creating a PR that changes this for all the targets?

jihlein, 2015-04-17:
That was my next step, once I heard back from some other platform users.  I am curious as to why the OSX binary changes. The Windows binary runs, I'm seeing a reduction in CPU usage and an increase in available heap space.

I'll get the pull request put together tonight.

peabody124, 2015-04-17:
Thanks for figuring this out. I know it took a while and lots of crashes. It's quite odd you might want to throw those lines above into your code before and after and see if the value of ```CORTEX_VTOR_INIT``` changes for you. It might be some later change that is just a crash from GCC on Windows.

jihlein, 2015-04-17:
Can the PR span both Quanton and Colibri targets?  I can't verify the Colibri build.

peabody124, 2015-04-17:
Sure. Anything required for "all_flight" to compile for you. Then we can validate the PR before merging it.

mlyle, 2015-08-01:
Fixed by #1560

closed 2015-08-01 by mlyle