TauLabs/TauLabs #1553: integrate HOTT SUMD Setup into wizard

Armadill0, 2015-04-07:
SUMD has been integrated with #820 into Taulabs, but it is still not available in the Setup Wizard.

Would be great if you could do this to make the setup easier with SUMD.

cGiesen, 2015-04-07:
It is!!!
1. Hardware: Select "HoTT SUMD" on one of your serial ports
2. Input: Select HoTT SUM for all types

By the way there is only SUMD left for Graupner. SUMH is gone

Armadill0, 2015-04-07:
Do we speak of the same things?

I'm speaking of the Tau Labs Setup Wizard:
[image]

cGiesen, 2015-04-07:
Sorry.
I never use the wizard ;)

Armadill0, 2015-04-07:
Today the same applies to me. But the wizard was one of the reasons why I and many of my fellows started using copters with Taulabs. :smile:

ernieift, 2015-04-15:
I also never used the wizard, but i can have a look at it.
@cGiesen my last GR-12SH was still shipped with SUMH ;).

ernieift, 2015-04-28:
I just tried the Configuration Wizard and it works.
Go to Configuration->Hardware, set a Uart to HoTT SUMx and reboot the controller.
Now go to Configuration->Input and select the "Start Configuration Wizard" and follow the instructions...
That's all.

ernieift, 2015-05-01:
After testing the wizard on CC3D, i can say, there is no possibility to select the wanted USART on this target. If you want to use the main port, you have to set this configuration manually. If you follow the guide above, everything works like expected.

peabody124, 2015-05-08:
@ernieift so can this be closed?

ernieift, 2015-05-08:
Because of the 2 HoTT protocols and 2..5 possible USART there are enough possibilities of configuration. I think it would be wrong if the wizard would put simply SUMD on a fixed (1st?) port. In addition the wizard works on the Quanton for none of the protocols (DSM, SBUS, etc.).
The ability to determine the port and the protocol in the hardware configuration and then to use the wizard, is sufficient.
For me, we can conclude here. What do you think @Armadill0 ?

Armadill0, 2015-05-11:
Sounds like a plan.

But I think if this part of the wizard doesn't work, it should be deactivated for the affected board. Maybe it could be replaced for all boards where this is not available/working with a hint (currently there's none if I understood aright), that this has to be done manually.

ernieift, 2015-05-11:
@Armadill0 feel free to try #1589. As said I, the quanton still hasn't got the radio wizard. There are too many possibilities for the USARTs. But if someone wants it, I could use USART2 for all possible protocols here. This doesn't prevents someone to setup the radio in the hardware tab. SUMH is for handmade configurations via hardware tab.

ernieift, 2015-05-27:
@Armadill0 can this be closed after merging #1589?

Armadill0, 2015-05-27:
Yes. :smile:

closed 2015-05-27 by Armadill0