TauLabs/TauLabs #1554: internationalization support for GCS

Armadill0, 2015-04-07:
With QT it is very easy to add support for different languages to a software written in it. As I have many friends who often say, that they don't want to use Taulabs because it is not available in their native language, I would recommend to integrate support for different languages to Taulabs.

A documentation for the integration is available here: http://doc.qt.io/qt-5/internationalization.html

To get a fast support of many languages I would recommend to setup a translation project on www.transifex.com for Taulabs. This translation community is very easy to use and there are thousands of users who translate several projects into other languages.
I'm using this for some of my own projects to get translations for languages, which I and other contributors don't speak ourselfs. And I was impressed how many translations have been approved in a very short time.
As Taulabs also is an opensource project there are ne fees to use Transifex. :smile:

peabody124, 2015-04-07:
Any chance you would be interested in spearheading this, if you have already worked with transifex? We should have most of the strings in the code already wrappered with tr()

Armadill0, 2015-04-07:
Sure, I'll have look into the project the next days to identify the next steps.