TauLabs/TauLabs #1555: Taulink settings do not populate

carbogit, 2015-04-12:
First, I am assuming that the OPlink Mini I have is treated as a Taulink. Using the Rf bind wizard, I have established a working link between Oplm and Revomini:

[image]

However, the settings for Hwtaulink do not populate, so I can't make manual changes without the wizard:

[image]

[image]

peabody124, 2015-04-13:
Yeah because of the interaction of the "SessionManaging" system the GCS doesn't know that HWTauLink exists and doesn't notice it. You can fetch the settings and it should show you the latest. Glad to hear the binding wizard is working for you.

carbogit, 2015-04-15:
Yes, bind wizard is working fine, a really nice solution.
Fetching settings does work (had GCS crash when doing that before, but not anymore), so problem solved.

One more thing, already posted on the forums. The coordID field does not accept the last digit in most hex IDs when working manually in the object browser.

1234567890 can be entered, but not
2358626261 which is my coordinator modem

peabody124, 2015-04-15:
Could you open a separate issue for that? It is a general bug in the UAVObjectBrowser that I would like to fix about limited digits. It doesn't handle negative numbers either IIRC.

mlyle, 2015-08-28:
As designed-- closing.

closed 2015-08-28 by mlyle