TauLabs/TauLabs #1572: BUG: Six point calibration can modify FC orientation relative to UAV

BrainFPV, 2015-05-05:
When performing the 6-point calibration, the GCS first fetches `AttitudeSetting.BoardRotation` and then sets all values to zero for the calibration. When the calibration is successful, the orignial values are written back to the FC. 

This is problematic, as the calibration can be interrupted, e.g. due to a USB disconnection when moving the UAV around during calibration, which will leave `AttitudeSetting.BoardRotation` set to all zero instead of the original values. When the calibration is attempted again without restarting the FC, the GCS will fetch the all-zero values and write them back to the FC after the calibration, which may be different from the values that the user originally entered. This can have bad consequences, e.g., if the FC is mounted facing backwards (Yaw=180), the UAV will perform a "flip of death" when spooling up.

The above also applies to the sensor scalings (`SensorSettings.AccelScale`, `SensorSettings.MagScale`). 

I see two potential ways to solve this:

1. Leave `AttitudeSetting.BoardRotation` unchanged and the GCS applies the inverse of the rotation/scaling matrix to the xyz sensor values during the calibration. 

2. Leave `AttitudeSetting.BoardRotation` unchanged and add a special mode to the Sensor module, which causes it not to apply scaling/rotation to the sensor values. This mode is only used for calibration and the Sensor module will remain in error state as long as it is enabled.

IMO, solution 2 may be better, as 1 needs us to replicate the calibration from the FC side in the GCS and if there is a discrepancy, the resulting coefficients will be wrong. Thoughts?

peabody124, 2015-05-06:
I think there might be an option #3. Six point calibration only looks at the vector magnitudes. A rotation matrix should not alter that. I think entirely ignoring board rotation while collecting data would be ok. After calculating the correction an inverse rotation would be requires though.

BrainFPV, 2015-05-06:
@peabody124 sounds good. I agree that it makes sense to do as much as possible in the GCS to not increase the complexity of the firmware. I will look into it.

PR #1578 (2015-06-08): FIX: don't modify board rotation during calibration
Fixes #1572

Board rotation is applied in GCS instead of resetting the rotation on the FC. Scales and bias are still reset, but this is much less critical.

Ready for review.
closed 2015-06-08 by peabody124