TauLabs/TauLabs #1590: deriv_gamma=0 in PID for "Luxfloat"

fujin, 2015-05-11:
Yo,

I was speaking briefly to Lux on IRC and he more or less mentioned that our PID looks the same, besides a recommendation to configure deriv_gamma=0. I'm going to do some test runs and will report back, any thoughts? Is it worth having a toggle to change this somewhere? 

Is there an equivalent static float in MWRate or is it completely different?

Thanks in advance.

fujin, 2015-05-11:
https://github.com/TauLabs/TauLabs/blob/4176d4895a7c9f164c3056c99291cf2ba3b702d0/flight/Libraries/math/pid.c#L150

fujin, 2015-05-11:
```
<Lux> i mean the default taulabs pid works the same way as luxfloat
<Lux> if you set deriv_gamma to zero the taulabs pid should behave the same as luxfloat
```

hope 'e doesn't mind being quoted here. :smirk_cat:

fujin, 2015-05-12:
nb: I am a mathematical moron so have literally no idea what is going on here

peabody124, 2015-05-12:
You can already do this by setting ```StabilizationSettings.DerivativeGamma```  to 0 in the browser.

fujin, 2015-05-12:
Thanks heaps @peabody124 will test and report back!

closed 2015-05-12 by fujin