TauLabs/TauLabs #1597: MotorsSpinWhileArmed

EjectSteve, 2015-05-12:
Using build taulabs_next_20150510_191255_be71c3afc6_win32.exe

With template gemini-prerelease-m2.uav

The MotorsSpinWhileArmed setting can be changed to FALSE, but returns to TRUE when I close GCS and go back in.  I have been sure to press send and save many times and have powercycled the FC as well.  This is a TBS Gemini.

Change motor outputs to oneshot works fine and the settings stay saved.

EjectSteve, 2015-05-12:
I changed the neutral settings  for motors to 1650 and also noticed and unchecked the option on the motor page that says "Motors spin at neutral output when armed and throttle below zero".  I guess this translates to "Motors Spin when Armed".  When I changed it on the motors page it also changed to FALSE for the "MotorsSpinWhileArmed" on the System page.  If this is the expected behavior, than for me, the issue is resolved.

SystemsGuy, 2015-05-20:
@EjectSteve did you clean up that uav before importing it?  Not 100% sure it's related, but there are a lot of old objects in it.

EjectSteve, 2015-05-20:
I did not clean the file as I'm not sure how that would be accomplished.  As far as I know, this build is working as expected since checking/unchecking this works: "Motors spin at neutral output when armed and throttle below zero".    Directly setting true/false on this setting does not save: MotorsSpinWhileArmed.  

Perhaps this is just an anomaly if the release I used.

mlyle, 2015-05-20:
One thing I've noticed is that directly changing booleans/enums in the uav object browser requires clicking on something else before sending to the flight board and saving.  Could this be related?

JosephRDawson, 2015-06-09:
I have the same issue. I have tried changing other values and they do save but it appears none of my booleans can be modified anymore. (Not sure if I ever could modify them because I never have.)

JosephRDawson, 2015-06-09:
I had to click on the logic variable and have it highlighted while i clicked save for it to work! How crazy!

mlyle, 2015-08-13:
Yah, UAVO save behavior is weird.

closed 2015-08-13 by mlyle