TauLabs/TauLabs #1617: UBX should reconfigure while "NoGps" or if messages missing

peabody124, 2015-05-16:
If you apply power via USB before the GPS or disconnect/reconnect the GPS then you will not have a configuration. The system should repeatedly send the autoconfiguration messages (e.g. once every 10 seconds) if the status is NoGps. possibly also if some messages (e.g. NEDVEL) are never seen.

DTFUHF, 2015-05-16:
Probably better idea is a simple state machine that can detect and recover
from configuration and communication problems. That can also handle
autobauding. Simpler FCs like baseflight already do this.

peabody124, 2015-05-16:
We already have autobauding. It is just deciding when to call it.

DTFUHF, 2015-05-17:
Yes, I meant a GPS communication state machine could handle this as well.

elafargue, 2015-06-08:
This is fixed by #1641

closed 2015-06-08 by elafargue