TauLabs/TauLabs #1619: ManualControl crashes on CC3D

ernieift, 2015-05-17:
Just found an issue on CopterControl. There is a
```
#define PIOS_MANUAL_STACK_SIZE          600
```
in pios_config.h. After the last merges the target crashes when the receiver should go on green. My quick workaround was this.
```
#define PIOS_MANUAL_STACK_SIZE          620
```
I tested the protocols HOTT_SUMD/SUMH. Eventually DSM has the same issue. This only happens on CopterControl. Other targets uses 1000 bytes stack. Before I make an PR for this I would prefer a more deeper analysis.

peabody124, 2015-05-17:
I would recommend a PR that increaes it a whole 100. It isn't useful to be so close, and gradually CC3D is just going to loose features. We really should recommend people just stick with the December release.

ernieift, 2015-05-18:
Solved by increasing stack on CC3D.

closed 2015-05-18 by ernieift