TauLabs/TauLabs #1627: Arming with Position Hold mode in flying modes

brat002, 2015-05-19:
Could you implement next feature:
Switch logic of Position Hold mode to Altitude hold logic, if no sat lock was acquired and back, if sat lock acquiring back. It will be very usable, when you are changing location too fast from open space to indoor without reconfiguring flight controller.

lolorc, 2015-06-03:
would setting profiles be a suitable workaround ?
 * acro profile (acro pids, no nav)
 * fpv profile (fpv pids, no nav)
 * nav profile (nav pids, nav)
 * ...
with a remote way to switch between them when not armed.

brat002, 2015-06-03:
Yes, may be, however fallback to altitude hold looks much simplier. We can add new flight mode "Mixed Position Hold", that will fallback to altitude hold, when GPS sat lock is unavailable by any reasons.

brat002, 2015-06-03:
Main goal of this feature - simple one-time configuration and many-time use In different conditions (indoor, outdoor, below, bridges, etc.). Where GPS signal will be available, system will have position hold. Where are GPS signal unavailable - system will use altitude hold.