TauLabs/TauLabs #1637: Oneshot: Issues in GCS after newest modifications in https://github.com/TauLabs/TauLabs/pull/1615

DTFUHF, 2015-05-25:
On the Output config widget:

1) The tooltip for the Type: field, refers to HPWM, which is not a selectable option.

2) The Max field is resized by GCS to 3 digits and the change can't be reversed. Steps to reproduce:

-Set an output channel to update rate = 2000 or higher when Resultion is at either setting of 1us or 80ns.
-Click Save.

3) Timer math doesn't appear to be correct. Example: 12kHz, 80ns resolution. For a 0-100% duty cycle output e.g. a brushed motor the min should be 0 and the max should be approx. 1041 (1041 * 80ns = 0.08328ms or about 12kHz) but this value can't be set higher than 833 in the GCS menu. It can be set higher in the UAVO menu, however.

Also, the GCS allows the setting of nonsensical values: for example, 1kHz / 1us / 9000 count max. I believe these are limited in actual flight code to 100% duty cycle values but it makes sense to also limit them here and show the user the proper value for max duty cycle.

ernieift, 2015-06-02:
Let me recalculate this. 12kHz PWM make 83,33...µs period. The maximum pulse width in integer are 84µs then.
You are right the 80ns are not quite correct. 83ns would be closer, but this is a small mistake.

mlyle, 2015-08-31:
#1642 corrected this.

closed 2015-08-31 by mlyle