TauLabs/TauLabs #1643: OPMap: map config contains a default home location, which interferes with actual UAV Home location

elafargue, 2015-05-27:
The OPMap gadget configuration contains a default home latitude and longitude. This is also used to set the map's Home location icon (the little house) from the configuration when the map configuration is loaded. 

Setting the home location on the map from the map's settings should be avoided, the map gadget should only use the home location that is present in the UAVObjects.

Proposed action: stop setting the map home location from map's default latitude/longitude

elafargue, 2015-06-02:
Any further comment on this ? I'd like to push a PR on this and close the issue if no one objects

mlyle, 2015-06-03:
I'm fine with this.

mlyle, 2015-08-13:
Merged in #1639

closed 2015-08-13 by mlyle