TauLabs/TauLabs #1644: OPMap: waypoint editor features overlap with PathPlanner gadget

elafargue, 2015-05-27:
Currently, it is possible to open a waypoint editor in a popup window. This does not follow the general UX guidelines of the GCS where we avoid popups as much as possible in favour of working on the workspace directly. Moreover, the popup window is identical to the Path Planner gadget, so I consider it as redundant.

Double-clicking on a waypoint opens a waypoint edit window, but changing the waypoint characteristics does not work (nothing is saved). This is also redundant with the Path Planner gadget waypoint edit capabilities.

Proposal:
- remove waypoint editor option from OPMap in favor of only editing waypoints in Path Planner gadget
- remove waypoint edit dialog from OPMap and have the Path Planner gadget be the only location where waypoints are modified in that way.
- Preserve the waypoint add features in OPMap, as well as modify waypoints via drag and drop since it makes complete sense.

mlyle, 2015-05-27:
As long as we can move and add waypoints on the opmap, I'm happy.  This sounds good to me.

peabody124, 2015-05-27:
I'm a little less convinced. The ability to modify a waypoint (altitude, speed, path segment type) from the map is pretty useful when making paths. Given that it isn't broken or badly structured, I'm a little hesitant to trash it. The way it is designed is quite nice because it all calls into the path model object.

elafargue, 2015-05-27:
Actually... It is completely broken, any edits from opmap are lost when you
click 'OK'.

At the very least, we should fix that...

(phone)

elafargue, 2015-05-27:
Example of GCS layout with map & path planner together: IMHO this makes editing waypoints super easy/fast and less error-prone since you can more easily spot a WP where altitude is not correct compared to the others...

[image]

mlyle, 2015-06-11:
Yah, I'm not a super big fan of things on the opmap pulldown, not least of which is that it sometimes behaves oddly when you right click on it.  That layout is nice.

It would be nice if the waypoints would be obviously marked "dirty" / not saved to board, too, because I've messed that up before.