TauLabs/TauLabs #1647: Expo in Advanced Stabilization

fujin, 2015-05-30:
Hi,

I think it would be cool to have an expo box in advanced stabilization tab beneath the degrees/sec value for pitch/roll/yaw.

Coming from previous flight controllers, I pretty much decided how much I wanted to roll (known in c.f. as RC rate, a command multiplier) and then reduced sensitivity with p/r/y rates (stick attenuation).

I was able to replicate the in-FC feeling that I am accustomed to by setting my p/r to 720deg/sec (two rolls per second) and y to 360deg/sec (one per) then setting P/R stabilizationsettings expo to 45%, Y expo to 70%.

cheers,

--aj

Trex4Git, 2015-06-04:
As i am just beginning with development for taulabs and have no experience with QT GUI development yet, only an early gui test: 
[image]

Trex4Git, 2015-06-05:
A small update. the expo settings located close to the range limits, so little bit more compact.
But i am not shure how the interest of others is in it, so i did not make a pull request yet.

[image]

fujin, 2015-06-05:
Nice, the 2nd definitely looks more compact.

Trex4Git, 2015-06-07:
Just to learn a little bit more about GCS coding and QT GUI development i implemented a test version of dynamic plots, that shows the combination of the expo value together with the full stick value on y-axis. And the normalized stick input on x-axis. (Layout got rearranged, line width, color etc only a early test; so still work to do, just showing the principal idea).

Plots are dynamically changed when the values in the boxes are changed.
Not compact anymore, but may be useful.

But still no feedback from the main developers on this topic.

[image]

Trex4Git, 2015-06-08:
A small optical update and fixed rearranged Layout issue.
[image]

I am interested in opinions regarding the design. Which is preferred, and does someone has a good idea for another layout?
The shown ones are just my first basic ideas, because of easy integration.

Trex4Git, 2015-06-09:
Another quick mockup for a layout idea:

[image]

fujin, 2015-06-11:
Might be starting to make sense to group the inner / outer loop stuff together (on two panes), one graph per panel? Inner loop (rate stab coeffiecient inner loop, full stick rate deg/s, expo rate, graph), Outer loop (atti stab c.o outer loop, full stick angle deg, expo horizon, curve)

I like the most recent mockup with the larger graphs but there is a lot of wasted space.

mlyle, 2015-06-11:
I like this a whole lot, and would totally vote for your pull.

I have a slight preference for the narrower layouts (the top 2) that show the graphs, because I think we might not work well on smal lscreens anymore.

Also I'd prefer it if the domain of the graph was from -1 to 1, just to make it more clear what's going on.

mlyle, 2015-06-11:
Please when you update the code put a comment into the flight side where the expo is pointing out that there's GCS code that graphs it that has to line up, and where.

mlyle, 2015-06-11:
Alternatively stick all 3 configuration panes on the left, and just the graphs on the right, and make the graphs narrower.

Trex4Git, 2015-06-11:
Thanks for your feedback.
I was not satisfied with any of my solutions above.
When i will have some  time tomorrow or on the weekend, i test some new layout and post them here for discussion.

Trex4Git, 2015-06-12:
Added line marks to horizon plot to show  for horizon the position where that maximally transitions to rate mode stars.
Otherwise every seems fine, maybe already final layout before PR?

[image]

Trex4Git, 2015-06-14:
Hi,
when starting to implement this, i just saw the 2 expo settings in the uavobject, and made them available in GCS. For inner loop / rate mode i think is easy to understand. 
But as the horizon mode is more complicated especially the plot might be misleading, as there is show not much data which is actually really applied in any mode, as discussed with icee/mlyle on IRC.
* In attitude mode no expo is applied, the curve shows only usable data for normal attitude mode when horizon expo is set to 0.
* In horizon mode it gets more  complicated, so at first i want to describe how i understand horizon mode from the sources:
   * For stick inputs above a threshold its purely a rate mode. 
   * For stick inputs below the threshold it blends in principal with a linear mix from attitude mode to rate mode. But there a some things that makes it difficult to describe the real behavior in this range:
      * the factor for blending from attitude into rate mode is not calculated for each stick input individual. The maximum of all stick input is used. So the control behavior for each axis is not individual, but depends dynamically on the other stick inputs.
      * the linear mixing between attitude and rate applies to the stick inputs after the expo is applied. When expo is not zero it is not linear for the raw stick inputs, only linear for the expo scaled stick values. So the horizon expo setting adds also a expo curve for the transition from attitude to rate.
      * the horizon expo setting is not only used for the horizon attitude curve and the transition from attitude to rate, but also for scaling the curve for the "horizon - rate mode". It seems to me that the "Full stick rate" is taken from the normal rate mode (inner loop), but for expo the horizon expo is applied and not the rate expo.

As this is a bit complicated, especially for new users, one solution may be removing the the horizon expo settings and plot from the stabilization widget

A other solution might be adding more information to the plot, maybe additional tool-tips and more information in the wiki.

I just started now with the plot:
* For  each axis there are now 2 curves. One for the attitude part (horizon angle) and in darker color for the rate part (horizon rate).
* The curves are scaled both with horizon expo (as i understand the source code).
* There are markers on the curves which show the threshold point. 
   * For stick inputs above its purely a rate mode (horizon rate curve) with expo scaling by the horizon expo. 
   * For stick inputs below there is the dynamic mixing between attitude and rate. In the example screenshot you can see that for different horizon expo settings the threshold value changes its position regarding the  normalized stick input on x-axis. 

I  think its not as easy to understand this plot as the plot for the inner loop, but advanced users should can get information from it that can help tuning the horizon mode. 
And as this is in the advanced tab, i think it might be OK to keep the horizon expo settings and plot, especially when additional information might be added.

[image]

Trex4Git, 2015-06-15:
No curves can be disabled/enabled with a click on each legend position. Maybe useful with the 6 curves in the horizon plot.

[image]

Trex4Git, 2015-06-17:
Started a PR: https://github.com/TauLabs/TauLabs/pull/1676

And also a new idea for a next PR with fixing/changing some of the horizon expo coupling.

Trex4Git, 2015-06-22:
PR is merged, should this issue now be closed, or wait for further feedback ?

mlyle, 2015-06-22:
Fixed by #1676

closed 2015-06-22 by mlyle