TauLabs/TauLabs #1653: CameraStab Save Error

jihlein, 2015-06-05:
After enabling the camera stabilization module, and applying and saving the changes, the apply and save buttons are filled with a yellow caution sign after applying and saving the changes.  Attempting to leave the page throws a changes not saved error message.

Resetting the board or recycling the power at this point and then returning to the camera page shows that the settings were saved and the camera stab module is no indeed enabled.

mlyle, 2015-08-21:
Could not reproduce in simulator.  Will try to try it on hardware, but deprioritizing.

jihlein, 2015-08-22:
I can still reproduce it with a build from todays next branch.  Here's  screen shot after clicking enable and then clicking save:

[image]

Hovering over the save button reads: Objects not present on the hardware: CameraStabSettings.  In the object settings, CameraStabSettings are indeed grayed out.

After a power cycle the error clears and the objects for camera stab appear under settings as one would expect.

Looks like a race of some sort.  Bad part is I've seen this error lock up the board to the point where the flash has to be erased to recover it.  This test did not lock up for what it's worth.  The lockup could be due to whatever the contents of the flash is, or isn't.

Unfortunately I'm totally ignorant with the GCS programming, or I'd try to fix it.....

mlyle, 2015-08-24:
I saw this on hardware when testing.  It's still tagged for this release but the work is stacking up and we're near the end, so I don't know if I'll get to investigate.

mlyle, 2015-08-31:
Sorry.