TauLabs/TauLabs #1663: Sparky2 reboots / USB telemetry crashes

elafargue, 2015-06-08:
On current 'next' branch (14e72bb8c56a) , on sparky2 target with no modules enabled, connecting/disconnecting USB while a battery is connected leads to either:

- Sparky2 reboot
- Telemetry in strange state: no settings objects can be retrieved, but regular uavobjects are still sent on the uav link. This feels like the telemetry receive task is crashed on sparky2 ?

elafargue, 2015-06-08:
It might be linked to the fact telemetry is also sent over the radio link - in any case, the behaviour is very confusing...

BrainFPV, 2015-06-14:
The strange telemetry state could be related to #1418. I have seen this happening when I have breakpoints in the firmware, so I assume it is caused by some sort of timeout in the GCS.