TauLabs/TauLabs #1672: Graupner PPM: only usable for <12 channels

beckerth, 2015-06-14:
When using a Graupner Hott Receiver GR-16 and activating more than 12 channels the RC input is not recognized any more.

tracernz, 2015-06-14:
The max PPM channel limit is set to 12 e.g. [Sparky2](https://github.com/TauLabs/TauLabs/blob/next/flight/targets/sparky2/board-info/pios_board.h#L200). Jeti supports either 8 or 16 channels on PPM with the latest firmware so perhaps this should be revisited, although perhaps there is a good reason for the 12 channel limit?

mlyle, 2015-06-15:
What do you ever do with all those channels? ;) roll/pitch/yaw/throttle/flightmode/arming/camerar/camerap/cameray/accdesired1/accdesired2 just gets me to 11..

If there's a decent use case for all the channels I'm glad to look through pios to see why the limit is there.

tracernz, 2015-06-15:
The problem with the Jeti implementation at least is that you can either have 8 or 16, nothing between.

mlyle, 2015-06-15:
ah, that makes sense.  I will try and look at this at some point.

beckerth, 2015-06-15:
12 channels should be fine. I found nothing about that in the wiki and it took me some hours to find out why the receiver wasn't recognized... 
So an update of the wiki page will more or less close the issue?! https://github.com/TauLabs/TauLabs/wiki/User-Guide:-Connect-Receiver

ernieift, 2015-06-16:
Why not use SUMD for this problem? SUMD can handle up to 32 channels and all channels are transferred within one frame in 11 or 22ms. PPM and 12 channels take more than 12*2ms = 24ms at least.

mlyle, 2015-08-03:
Duplicate #818

closed 2015-08-03 by mlyle