TauLabs/TauLabs #1687: Rapidflash/OpenLRSng/BF/CF Chrome-style "configurator"

fujin, 2015-06-24:
So the GCS is kind of unwieldy, maintaining it for multiple dev environments results in a lot of time suck. I've contemplated setting up my own continuous build / delivery infrastructure to help with my own (and nominated) testing. I tend to get all of my boards (complicated now by having a mixed brain/sparky2 fleet) running the same revision or near enough that GCS doesn't puke, then hope like hell that I don't have to flash anything at the field.

I get there is a link time? coupling to UAVOs between FW/GCS, which could make this more complicated, which may involve some type of dynamic UAVOs upgrade (where compatible? I don't know.)

Would it be possible to have a Chrome-style GCS configurator? Are there any other benefits here I've not perceived, potentially for code re-use onto the Android platform?

cheers!

tracernz, 2015-06-24:
Chrome provides serial, USB, and USB-HID APIs so it's certainly possible. You could use the UAVO XML files easily enough in Javascript. I've thought a bit about doing this to make a light GCS targeted at acro flyers (i.e. 250 mini-quads). I think it would certainly help make TauLabs a lot more attractive to that group.

VArcht, 2015-06-25:
I'd rather it not be chrome app. Hard problem talking to the fc though.
 Couple ideas are:
 *Running a local server, basicly wrap the UAVObjects in a http server and api.

tracernz, 2015-06-25:
The app can communicate directly with the FC using the Chrome USB or serial APIs (https://developer.chrome.com/apps/app_usb). ;)

dustin, 2015-06-26:
I rather like the chrome app idea.  It would require solving the "too many versions" problem, but the advantages are pretty great in other systems where I use this:

* It updates automatically.
* It runs on my chromebook (and other computers) without effort.
* It doesn't have rendering artifacts I experience with the existing GUIs.

I'm sure it's a lot of work, but much of that might just be creating the One True GCS concept.