TauLabs/TauLabs #1689: WS2812 addressable LED support

fujin, 2015-06-25:
It would be rad if we could control WS2812 addressable LEDs like http://multirotormania.com/led-lighting/989-mrm-cleanflight-5050-ws2812-programmable-led-bar-70mm.html or http://multirotormania.com/led-lighting/1018-cleanflight-5050-sw2812-programmable-led-.html

Cleanflight has a neat GUI for this: https://youtu.be/SdN_4YaaxaU?t=9m13s

Thoughts?

cheers!

dustin, 2015-06-25:
It should be doable in picoc if at all.

Adding flight controller support for operating a light is silly.  Adding flight controller support for allowing users to operate things you haven't heard of is grand.

fujin, 2015-06-25:
@dustin what about GCS support for providing inputs to some "core plugins" picoC? I agree that doing it in/on the FC beyond that is probably not the right choice.

dustin, 2015-06-25:
Yeah, it'd be neat.  the ws2812 protocol needs a bit of care to get right, so not having to do that every time would be good.  I think it could be an example for what kinds of things would be useful for distributing scripts (even if it ends up being just paste a big wad of gist in).

lolorc, 2015-06-25:
I'm using the method described here: https://github.com/TauLabs/TauLabs/wiki/PicoC-Project:-RGB-LEDs
works ok.
We need some kind of repository to share picoC scripts tho.

Trex4Git, 2015-06-25:
Just wanted to post the same link ;)

But what is not mentioned there. I read that now you can also use I2C to communicate with the Attiny, that drives the led. With that instead of UART connection it should be also possible to connect stripes in parallel.

And also i have seen some more advanced skripts compared to that shown in the wiki, i think it was there:  http://fpv-community.de/showthread.php?55909-TauLabs-PicoC-Ansteuerung-RGB-LEDs-WS2812&p=735768&viewfull=1#post735768

lolorc, 2015-06-25:
yep i2c also works ok. The only issue I have with it, you can't leave your attiny connected when only using usb to connect your  FC, it will panic, attiny has to be powered otherwise the i2c_usart stuff will fail to initialize and panic.
see https://github.com/TauLabs/TauLabs/blob/next/flight/targets/quanton/fw/pios_board.c#L861 for example.

ernieift, 2015-07-04:
I2C is cool. You have access to the LED buffer like an eprom (24LC01) now. It is also possible to put some ATtinys in parallel and address them as separate devices.
The actual version is here:
https://github.com/ernieift/NeoPixel_Transceiver
I think everyone with Experience in Arduino (MWii and MavLink) is able to use this. It it far more flexible than using an LED configurator. There are always things which aren't represented. Think of Kight Rider (KITT), Landing Lights or Brightness control depending on the distance...

dustin, 2015-10-23:
This would be much easier/brighter/harder/faster/etc...: https://blog.adafruit.com/2015/10/21/new-product-pixie-3w-chainable-smart-led-pixel/

tracernz, 2015-10-23:
@dustin Those look much nicer to support! If I read it correctly they speak 115200 baud serial and only require an update every two seconds, although every 1 second is recommended.

mlyle, 2015-10-23:
have to update them many times per second to animate ;)