TauLabs/TauLabs #1690: GCS Module: 1wire protocol 4w-if SK/BL ESC flash, settings

fujin, 2015-06-25:
It may be viable to form a partnership with either the team behind Rapidflash, or the BLHeliSuite team, and implement a GCS pane for ESC configuration. If the configuration/flash protocol are open, or simple enough, such a partnership might not even be required, but some consideration must be given to the delta between upstream updates / new features and the TL UI lagging behind.

I have no doubt that we would be able to do enough to flash and re-configure ESCs with no change to wiring required.

Thoughts?

cheers

fujin, 2015-06-25:
https://github.com/TauLabs/TauLabs/issues/1687 -- rapidflash Chrome configurator reference may come in handy.

fujin, 2015-06-25:
@Brotronics kindly pointed me at https://github.com/DTFUHF/baseflight/blob/usblinker/src/usblinker.c

dustin, 2015-06-25:
I would love this.  I've had a couple builds where getting to ESCs is just hard enough to make this unpleasant.  Not only for changing direction, but also experimental reconfiguration (e.g. comp_pwm) and builds.

peabody124, 2015-06-25:
I also think this would be an awesome feature. As a suggested implementation method, I would write a PIOS_COM interface that bit bangs the serial protocol. I did something very similar in my ESC code using a soft USART that sat on the PWM lines: https://github.com/peabody124/TauLabs/blob/esc/flight/PiOS/STM32F10x/pios_softusart.c

fujin, 2015-06-29:
https://www.youtube.com/watch?v=2N6L8Igcgc4

fujin, 2015-07-28:
https://github.com/nathantsoi/cleanflight/blob/serial1wire/docs/1wire.md

dustin, 2015-07-28:
IMO, the Cleanflight way is a really dumb cop-out solution.  OpenPilot did a much better job (though there's more work to do) and I think Tau Labs can do even better.

I want "install image on all my ESCs" and potentially "motor 3 is backwards."  Options, presumably, should be consistent across all of them.

Using my flight controller as a serial adaptor feels very lame and platform specific.

Trex4Git, 2015-07-28:
I think for the CF way much less  work would be needed. 
As BlHeli is also under active development, and a lot of work is already done in BlHeliSuite especially as a first step implementing an "interface passthrough" in Taulabs should be fine.

Otherwise it would be a lot more work to get the BlHeliSuite functionality into GCS, and also someone must do the maintaining when there are changes in new BlHehli Versions.

dustin, 2015-07-28:
Doesn't OP's work on more than just Windows?  Because this
proposal/cleanflight's is Windows only.