TauLabs/TauLabs #1698: Include automatic bootloader update in firmware

BrainFPV, 2015-06-29:
I just added some code to include the bootloader image into the normal firmware. On boot, the firmware computes the CRC32 of the actual bootloader and the bootloader image included in the firmware and updates the bootloader if necessary. This is useful for making sure all users are running the latest bootloader version.

The commit below enables this for the Brain, it can easily be added to other targets as well. Thoughts?

https://github.com/BrainFPV/TauLabs/commit/61195565733c19b32744a03d7b913575958167dc

fujin, 2015-06-29:
Automatically upgrading the bootloader seems like a safe enough thing to handle, what benefits are there or potential breakages? is there a chance the bootloader will fail while writing and render the board bricked until you dfu/bootpad?

I've never upgraded a bl on my brain or tl boards before (I didn't even realize you had to do it separately)

fujin, 2015-06-29:
from irc:

```
<varcht|> colibri and quanton need to since jan, I like it hshred       [11:06]
<hshred> early versions of the brain have an old bl which doesn't support some
         of the features and the partition sizes are wrong
```

Trex4Git, 2015-06-30:
When i had to upgrade the bootloader on my Quanton to use latest next versions, i had to read and ask some small things.

I think inexperienced users may don't feel comfortable with this.
So i think this would help.
 
And as it already auto detects if the bootloader must be flashed or not, maybe an additional prompt to ask the user whether he wants to upgrade or not may be useful.

mlyle, 2015-07-04:
I think this should look at configuration and default off for safety, since it can brick things and require dfu to fix.

peabody124, 2015-07-04:
I also vote against this by default, but it is a great feature when a bl update is required

BrainFPV, 2015-07-08:
I'm closing this. I agree that it is best to have this feature disabled by default (which it is, it only includes the BL update if `EMBED_BL_UPDATE=YES` is set). I will use this for the Brain with the upcoming release to make sure all users have the same BL version.

closed 2015-07-08 by BrainFPV

mlyle, 2015-07-09:
Here's a question-- is it feasible to get rid of our bu_ images and instead produce a different image with this enabled?  At least then you wouldn't have to flash bootupdate, reboot, reboot, go to bootloader, flash real image, etc.