TauLabs/TauLabs #1709: Issues with next TL7_5_15

pug398, 2015-07-06:
PFD does not report # of satellites correctly. Instead of numbers has strange characters. Started last few releases and config files deleted and reinstalled. Opengl32.dll installed and graphics have been working fine. Windows error on closing of GCS appears fixed.

mlyle, 2015-07-06:
Yes.  I believe this is a bug / limitation in QML that QT 5.5.0 may be expected to fix.  Before we had two PFD implementations-- the C implementation and the javascript/qml future one;  I removed the old one and there are definitely some regressions / limitations in functionality there because it seems QML can't get to some types.

pug398, 2015-07-06:
Ok Mike. Just wanted to make sure that it was known and didn't get overlooked before going to release. Looks like the other issues I ran into have been resolved.

mlyle, 2015-08-29:
Believed to be resolved.

closed 2015-08-29 by mlyle