TauLabs/TauLabs #1716: Strange tab input selection order in Stabilization / Advanced screen

dustin, 2015-07-13:
Tabbing between inputs, I get this really strange order:

[image]

Also, to even get there from the previous ones, it jumps through a couple odd places (system viewer on the right and module selection on the left).

Trex4Git, 2015-07-13:
That is good to understand, thanks for the feedback.

I hope that today after work i find the time to take a closer look at it, and i think fixing it would not be much work.

peabody124, 2015-07-13:
There is a button at the top of the designer that lets you edit tab order:
[image]

Also, when saving the .ui file, make sure to go back to the basic tab first. It preserves this as the default tab when you go there. It would be nice to have this fix in the final release.

Trex4Git, 2015-07-13:
Already had seen that this morning, and now had time to fix it.

[image]
[image]

And as not only the advanced tab with the new expo settings had strange tab order, i changed it in the complete stabilization widget:

[image]
[image]
[image]

When this is OK i can make a PR from this branch: https://github.com/Trex4Git/TauLabs/commit/a4351d6d479798b569e759e43b179e4067be4d81

mlyle, 2015-07-13:
Fixed by #1718

closed 2015-07-13 by mlyle