TauLabs/TauLabs #1722: UAVTalk over USB HID connection, written bytes end in read buffer

vitaminwater, 2015-07-15:
I have a weird bug happening, I'm working on a Golang implementation of the UAVTalk protocol, everything works fine except when I write to the USB connection, the controller does receive the data, because it does behave accordingly, but then the next read contains the last written bytes randomly placed in its buffer.

example:

I send:

`02:12:3c:20:11:00:4a:4e:03:89:00:00:00:00:00:00:00:00:00:15`

And the next read is:

`e7:e7:00:00:02:29:7e:2f:00:02:23:00:9a:3c:20:24:00:e6:d5:da:33:63:f9:f8:3e:b5:74:3b:3f:dd:bc:f0:3e:df:1a:a2:bd:e2:e5:00:43:76:02:12:3c:20:11:00:4a:4e:03:89:00:00:00:00:00:00:00:00:00:15:e7:e7:00:50`

(written packet start at the 24th position from end)

What makes me think that the problem might come the the flight controller is that if I send a wrongly built packet, eg. if I remove the crc8 from the packet (replace it by 0), the bug does not occure, which would be the case if the bug was on my side or in the USB HID library (hidapi) or in the Go wrapper (go.hid).

thanks.

guilhermito, 2015-07-15:
That received data looks strange the first two bytes should be the report ID and data length

vitaminwater, 2015-07-15:
No problem on this part, I copy pasted from my terminal, here's 2 complete packets:

sending:
`3c:20:1d:00:44:27:c7:ab:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:03:97`

receive:
`3c:20:24:00:e6:d5:da:33:98:86:cf:3c:be:fa:5e:3f:ba:d9:fa:3e:28:e5:d0:3c:34:09:30:43:a1:bd:02:1e:3c:20:1d:00:44`

vitaminwater, 2015-07-15:
oOps, I stripped it from my output... anyway this part is ok, the flight controller behaves accordingly on handshake and sessionManaging.

vitaminwater, 2015-07-15:
You can still see the report id and length in the received packet duplicated in the 7 last bytes.

vitaminwater, 2015-07-15:
could it be linked to #1418 ?

guilhermito, 2015-07-15:
I don't see what is the problem the packet reports 32 valid bytes and you receive 32 valid bytes

vitaminwater, 2015-07-15:
The problem is that if you look closely, the packet that I send is duplicated in the last 7 bytes of the next received packet, the length if good, the end of the packet is just a part of the last sent one duplicated.

So the crc is wrong, but anyway the packet is just garbage then.

here's a screenshot of what is going through (reportID and length are already stripped, but you can still see them is the duplicated bytes) http://imgur.com/Mm1luYN

As you can see, the first write goes ok (it goes through before anything else), but then everytime (nearly, it is quite random) I send something, part of it gets duplicated in the next packet that I read (where it says "WARN[0000] Wrong crc8 !!!!").

vitaminwater, 2015-07-15:
I can still go though Handshake and start doing the SessionManaging stuff (notice "HandshakeAck", "Connected", "Started noSession state" and "0 0 0 51 0" 51 being the number of active uavo).

sorry for screenshot, but at least there are colors to show the errors :/

vitaminwater, 2015-07-16:
After some more investigation with usbdump it appears that the bug is also present when I use the GCS.

`
0.105398 -->1 64: 021e 3c20 1d00 4427 c7ab 0000 0000 00a0  0944 0000 0000 0000 0000 0000 0000 01aa  0000 0000 0000 0000 0000 0000 0000 0000  0000 0000 0000 0000 0000 0000 0000 0000
  0.109013 1<-- 64: 0136 f63c 2024 00e6 d5da 33df 2e08 3ffc  e0b8 3e00 78f0 3c24 ee43 bf98 9dc5 41b8  021e 3c20 1d00 4427 c7ab 0000 0000 00a0  0944 0000 0000 0000 0000 0000 0000 01aa
`
as you can see, the written data is duplicated in the next write at the 33rd byte.

solidgoldbomb, 2015-07-16:
I'm assuming that you're running on linux (usbmon user).  What distribution + version are you running?

On your sparky, are you running a release version or a version you compiled yourself?  If you compiled it, could you tell us the git hash that your tree is at?

Could you capture the USB issue using wireshark and post the .pcap file somewhere so I can see the entire exchange?

vitaminwater, 2015-07-16:
I've had the bug with osx 10.10.4 and Archlinux just after a pacman -Syu (I'll post the uname -a when I'm back home, but it should be 4.0.7-1 as per https://aur.archlinux.org/packages/linux-ck/).

I'm using a compiled fw+gcs (didn't even know there was a downloadable version), last git log entry is:

`
commit 9effa7b30810ffcc9d85485f22ad42810abac941
Merge: 79c0c22 a73a95e
Author: Stacey Sheldon <stac@solidgoldbomb.org>
Date:   Tue Jul 14 15:30:00 2015 -0400

    Merge pull request #1719 from mlyle/mpl-removerevo
    
    revolution: remove base revolution target
`

I started recompiling a wireshark with the uavtalk plugin, still in progress, ended up using usbdump directly to have a quick glance. I'll post the pcap when I'm home.

solidgoldbomb, 2015-07-16:
You shouldn't actually need any plugins or a special build of wireshark.  The normal version should be able to capture the raw USB URBs.  It basically works the same way as usbdump but gives me a binary format that I can search/filter on.

Maybe also send me the output of ```lsusb -v``` after you plug in your sparky but just before you run the packet capture.

I'll also try to reproduce this on sparky2 (I don't have a sparky1) on the weekend but would like to have your capture ahead of that (if possible) so I can compare against what I'm seeing.

vitaminwater, 2015-07-17:
I was talking about recompiling wireshark after a quick chat I got with peabody and stac about the wireshark UAVTalk plugin, which could make the whole thing much more convenient.

You'll find the pcapng file here : http://ccsas.biz/Taulabs_GCS_USB.pcapng

and lsusb here https://gist.github.com/vitaminwater/46f84b0f23dfa483b8a0

solidgoldbomb, 2015-07-17:
(@solidgoldbomb == stac)

Thanks for posting the capture and lsusb output.  The contents of the capture look fine to me.  Maybe I'm missing something subtle.

When you're processing the HID reports, they will all contain exactly 64 bytes of raw data (that's the configured report size for the device).  The first byte will be the HID report ID (1 = Device->Host emulated serial channel for telemetry), the second byte will be the number of *valid* bytes contained in the 64 byte raw report buffer.

When you are processing the HID reports, you need to ignore the invalid bytes at the tail of the HID report since that's just random garbage.  Are you doing that in your golang implementation?

Also, just to be clear, the data stream received on HID report 1 is an emulated serial stream.  You shouldn't assume that each HID report will start with the Sync framing byte (0x3C).  Some UAVTalk messages will span across multiple HID reports if they are either larger than 64 bytes or if they happen to line up in the serial stream where they don't entirely fit at the end of one report and spill out into the next one.  This is entirely normal.  Were you handling this case in your golang implementation?

To see an example of how the HID report stream is handled in the GCS, take a look at ```RawHIDReadThread::run()``` in ```./ground/gcs/src/plugins/rawhid/rawhid.cpp```.  You can see it split off the report ID and valid count before handing the serial stream to the UAVTalk handler.

If you want to watch the details of the UAVTalk protocol in the GCS, you can recompile with ```#define UAVTALK_DEBUG``` uncommented at the top of ```./ground/gcs/src/plugins/uavtalk/uavtalk.cpp```.  That's one implementation of UAVTalk deserialization.

Hope this helps with your golang implementation.  If I'm misunderstanding your bug report or if you have any further questions, please let me know.

vitaminwater, 2015-07-17:
Thanks for your time, I surely still have misunderstandings as I've only started working on taulabs few months ago (spare time, you know what I mean.).
But unfortunately what you describe was well understood and implemented, and the files you link to were also read.

What I'm talking about can be seen in the 130, if you look at what packet 128 (sent from GCS) starts with:

`02 1e 3c 20 1d 00 44 27 [...]` (1)

You will see at packet 130 starting with:

`01 3b 95 3c 20 24 00 e6 [...]` (2)

that at byte 96, packet 128 is repeating itself (line 0060 of packet 130 in wireshark). The problem is that this packet is meant to be 0x0024 (36) and the bytes of packet 128 start at position 30 (from 0x3c) of this packet.

Actually you can see this in all packets "in" following a "out", it's not always an issue though, as some packets are shorter than where the duplicate starts.

(1) Why does the packets sniffed by wireshark contain 64 bytes before the actual packet ? (protocol encapsulation I guess ? URB, right ?).
There's also this 64 bytes packets inbetween (same reason I guess?)
(2) Why is the 3rd byte not 0x3c, I'm aware that packets could be split in multiple parts, but this is not the case as the previous "in" packet is just 0x001d (29) length

solidgoldbomb, 2015-07-17:
Ok, I think I see what you mean in packet 128 and 130.  Will investigate further.  Thanks for clarifying.

vitaminwater, 2015-07-17:
I just made an abstraction layer for the flight controller link in my implementation, and the bug is not there.

vitaminwater, 2015-10-13:
Hi, just a little update, I just got a sparky2, and the bug does not reproduce on this board:)

mlyle, 2015-10-13:
Ah, thanks for the note!  I hadn't noticed it was just sparky and I had tried peeking at this on a sparky2.  I'll try and make some time to check out the f3 implementation of the USB stuff.

vitaminwater, 2015-10-13:
Yes, looks like it's f3 only, I also had the bug on a discoveryf3.