TauLabs/TauLabs #1724: Attitude task doubles in CPU usage with high frequency (250hz) gyro lpf

fujin, 2015-07-16:
[image]

Reproduced on Brain-20150714. Dropping back to 184hz brings the cpu down to 7% on the Attitude task.

fujin, 2015-07-16:
@BrainFPV :ocean:

BrainFPV, 2015-07-17:
@fujin this is a limitation of the MPU9250; when the lpf is 250Hz, the MPU9250 is running at 8KHz, thus causing crazy CPU usage. I removed the 250Hz option for Brain, the same should probably be done for sparky2. 

I know you would probably like it, as it should give a super low tau in auto tune ;).

fujin, 2015-07-17:
when #1726 lands we can consider this resolved I think. We should maybe document the issue for future MPU9250 targets?

https://github.com/BrainFPV/TauLabs/commit/af84efa74130d1afdf08712206d8b4fef5520e0f

mlyle, 2015-08-01:
Fixed for now with #1726

closed 2015-08-01 by mlyle