TauLabs/TauLabs #1730: [WIP] Flight: adding gyro sub-samplng functionality

Trex4Git, 2015-07-18:
I have read somewhere on the forum, that it is preferred to disable the DLPF filter in the MPU an use the software IIR LPF filter (https://github.com/TauLabs/TauLabs/blob/next/flight/Modules/Stabilization/stabilization.c#L296-L298) instead.

But with the MPU9250 this is not possible, as the sample rate divider in the MPU only works when the DLPF is used (than the base frequency is 1kHz). When the  DLPF is disabled the output sample rate is 8Khz, what generates a high CPU load.
See https://github.com/TauLabs/TauLabs/pull/1726 and https://github.com/TauLabs/TauLabs/issues/1724

I have already thought about, and also read on IRC about, to run the MPU with no DLPF at 8kHz, but don't trigger our control loop with every interrupt, but do a sub-sampling on the MPU data, and run the control loop only every n interrupt.

That's why i already did a simple implementation: https://github.com/Trex4Git/TauLabs/commit/a88e63f64d66bd51c478c8ffabba813bae71dc4c?diff=split
The diff view seems to be a little bit strange. What I have done is that in the stabilization task directly after new gyro data is recognized, the data is read and the already existing gyro IIR lLPF filter is run,  and a counter ("gyroSubsample") is incremented. This data reading and filtering is run with the full MPU rate. 
But the rest of the stabilization task is run only every "GyroSubSampling" (a new UAV Object).

To test it, I set the outputs to SyncPWM (to check the control rate frequency with a logic analyzer). 
Reducing the control rate with a fixed MPU rate seems to work as expected, but the CPU load is higher then hoped. 

Tested with a Quanton. It has the MPU6000, but i don't have a FC with MPU9250, an as the MPU6000 also works with a sample rate divider in the MPU,no DLPF and 8Khz base frequency, I think its OK to do the test with it.

MPU rate | sub-sampling | control rate | CPU load
------------ | ------------- | ------------- | -------------
2 kHz | 1 | 2 kHz | 52%
2 kHz | 2 | 1 kHz | 46%
2 kHz | 3 | 666,67 Hz | 44%
2 kHz | 4 | 500 Hz | 43%
2 kHz | 5 | 400 Hz | 42.5%
2 kHz | 6 | 333,33 Hz | 42%
1 kHz | 1 | 1 kHz | 28%
500 Hz | 1 | 500 Hz | 16%

Especially when you compare the 1kHz & 500Hz control rate achieved with 2kHz MPU rate and a sub-sampling with the native reduced MPU rate, i think there is to much overhead with this implementation and room for improvements. 

So next thing i want to try, is running the filter in a earlier stage of the data processing (PIOS_MPUXXXX_Task, that seems to be triggered by the MPU interrupt function) , and let the stabilization task run only every "GyroSubSampling".

Trex4Git, 2015-07-18:
To change not to much at the same time, i want to use the already existing and for every target used IIR LPF filter. 
Improvements with a better filter can be done later when there is a working solution, i think.

Trex4Git, 2015-07-19:
Second implementation in a first version is done and tested: https://github.com/Trex4Git/TauLabs/commit/32e35d18bdc098c0ed94fb5a083744df166311bf
Some more details are discussed there: https://github.com/TauLabs/TauLabs/pull/1726#issuecomment-122561295

The sub-sampling is done now in the MPU task (with averaging the gyro data)  instead of the stabilization task.
Than the control loop is triggered with the averaged data and the lower frequency.

Only the gyro data is sampled with the higher frequency of the MPU rate and is than averaged,
accel and temperature data are sampled with the lower gyro sub-sampling frequency to reduce the CPU Load (and as accel data is also in the MPU internally slower sampled then the gyro data)

The sub-sampling setting is stored in the sensorsettings UAV Object, and from the sensors module provided to the PIOS MPU driver.

MPU rate | sub-sampling | control rate | CPU load
------------ | ------------- | ------------- | -------------
2 kHz | 1 | 2 kHz | 53%
2 kHz | 2 | 1 kHz | 35%
2 kHz | 4 | 500 Hz | 25%
1 kHz | 1 | 1 kHz | 28%
500 Hz | 1 | 500 Hz | 16%
4 kHz | 2 | 2 kHz | 65%
4 kHz | 4 | 1 kHz | 47%
8 kHz | 4 | 2 kHz | 90%
8 kHz | 8 | 1 kHz | 72%

Compared to the first implementation the CPU load is significantly reduced, but still higher at the same control rate when compared to native sampling with a reduced MPU rate = control rate (what is not possible for the MPU9250).
But with a higher MPU rate and sub-sampling more information is given to control system.

A further reduction of CPU load should be achieved, when the sub-sampling is again moved to a earlier stage of the data processing.
So I will also test doing the sub-sampling direct in the MPU interrupt function, and than doing the complete data reading and processing with the reduced frequency.
Than we have no advantage with more information because of the higher MPU rate, but CPU load should be reduced, and in principal this should work like using the sample rate divider in the MPU6000 with 8kHz base frequency.

Depending how this implementation will perform, maybe a combination sub-sampling stages could be useful.

Trex4Git, 2015-07-19:
Third implementation done and tested. https://github.com/Trex4Git/TauLabs/commit/aff75512f99632a355263d3740fddb97a2c08769

The sub-sampling is done in the MPU interrupt function with ignoring / not using some gyro data (specified by the sub-sampling), than everything else is not modified, but runs at the reduced frequency.

The sub-sampling setting is stored in the sensorsettings UAV Object,
and is provided from the sensors module to the PIOS MPU driver.

MPU rate | sub-sampling | control rate | CPU load
------------ | ------------- | ------------- | -------------
2 kHz | 1 | 2 kHz | 53%
2 kHz | 2 | 1 kHz | 28%
2 kHz | 4 | 500 Hz | 16%
1 kHz | 1 | 1 kHz | 28%
500 Hz | 1 | 500 Hz | 16%
4 kHz | 2 | 2 kHz | 53%
4 kHz | 4 | 1 kHz | 29%
8 kHz | 4 | 2 kHz | 54%
8 kHz | 8 | 1 kHz | 30%

As I had expected the CPU load is again significantly reduced and now only a small or no overhead compared to the native sample rate divider in the MPU6000 is noticeable.

So maybe a combination of the second and third implementation can be useful, e.g. at first reduce from 8kHz to 4kHz with efficient  MPU interrupt based sub-sampling, and than a further reduction to 1kHz or 2kHz by MPU task based sub-sampling.

Trex4Git, 2015-07-19:
Maybe labeling the the interrupt based as "Drop" and the task based as "AVG" or "LPF" and putting them in one UAV Object with 2 elements.

Trex4Git, 2015-07-19:
Did the combination of second and third implementation: https://github.com/Trex4Git/TauLabs/commits/flight_gyro_subsampling_v23
Also as discussed the sub-sampling setting is changed from the sensorsettings to the target hw UAV Object, and is now provided not from the sensors module but from pios_board.c to the PIOS MPU driver.

Just a quick test, but performance seems to be as expected.

MPU rate | IRQ sub-sampling  | Task sub-sampling | control rate | CPU load
------------ | ------------ | ------------- | ------------- | -------------
8 kHz | 2 | 4 | 1 kHz | 48%

mlyle, 2015-07-19:
So I tried to comment on this before (I'm on mobile) and it was lost, I think.

I do not believe an IIR is an appropriate way to downsample stuff.  We need to use an appropriate filter that will provide appropriate phase margin and an adequate stopband to prevent aliasing.  I think the filter as written is inferior to the mpu9250s internal filtering.

I also believe there are more appropriate ways to do this if we think there are gains available here. We have fifos in the mpu-9250 and do not need to wake up for every conversion.  We also have queues that could let us move multiple samples to later stages and integrate attitude estimates based on the individual samples.

Trex4Git, 2015-07-20:
We can do all these things and test them, to see the performance of them.

I can work on all these things in principal, but it will be  a little bit more work. So i would prefer doing them step by step, depending how much free time i have for this.

Trex4Git, 2015-07-20:
But we should also keep an eye on if its the possible advantages are woth thework that has to be spend on this.

With the third implementation you could also get some advantages compared the latest next.
- with the IRQ sub-sampling you can raise your control rate without to much overhead and should get lower latency in the system (I think in the autotune thread there where some results with 500Hz, 1kHz and 2kHz on the same setup, and 2kHz MPU rate produced the lowest tau)

- with the Task sub-sampling and simple averaging you should also already reduce some noise that is in the data

Things that should be investigated further forother implementations are:
- is there an advantage of the 32kHz MPU sample rate compared to 8kHz
- where should advanced filtering be done

mlyle, 2015-07-20:
So CIC may sound complicated, but it's really incredibly cheap.  A third order CIC to decimate by 1/24 looks like (laid out simply and procedurally for illustration):

    uint32_t integrateState[3]={};
    uint32_t derivativeState[3]={};
    uint32_t sampleNum=0;

    void HandleInput(uint16_t gyro) {
        integrateState[0] += gyro;
        integrateState[1] += integrateState[0];
        integrateState[2] += integrateState[1];

        sampleNum++;  if (sampleNum < 24) return;  // discard 23/24 samples.
        sampleNum = 0;

        derivativeState[0] = derivativeState[0] - integrateState[2];
        derivativeState[1] = derivativeState[1] - derivativeState[0];
        derivativeState[2] = derivativeState[2] - derivativeState[1];

        float output = derivativeState[2] / 13824.0f;      // 24 * 24 * 24
        Output(output);
    }

There's a lot of advantages to this: First, it has a better filter response than an IIR or fixed-bin FIR, and a relatively low delay.  (It rolls off 48dB, and infinitely at DC-- see figure 8 here: http://www.embedded.com/design/configurable-systems/4006446/Understanding-cascaded-integrator-comb-filters )

Second, it is all implemented in very cheap fixed point twos-complement math, and so can easily run on a 32KHz sample rate.

Third, it takes advantage of dither from sensor noise in a way that the internal FIR filters in the MPU-9250 don't.

I think the big win of this is that we can change from using the IRQ interrupt and drain samples at a 32KHz rate, and then have a configurable flight loop rate.  Then the sensor data will come in at close to nyquist.

How I would implement this is to turn on the FIFO on the MPU-9250, and set up a fixed task that drains samples from the FIFO and integrates them at 1.333KHz.  The delay between iterations could be controlled to get the same number of samples in each bunch (to discipline the task clock to the MPU-9250 sampling clock).  Then the output could be sent to the flight loop running at 1.333KHz.  The overhead would be minimal.  And this would be pretty much the optimal we could ever get from the hardware-- we'd not have to try many combinations and many possible tunings.

Trex4Git, 2015-07-20:
I already looked at the link you posted, but thank you. The link i posted regarding the advanced 2tab IIR LPF is from the same author (R. Lyons), and in my Link in Figure 4 this is also compared to a CIC filter.

When i understand your example correctly we should collect 24 samples in the mpu fifo, transfer them to µC and than the handle has to be called 24times with each sample as input. The integrate state is computed for every sample, but the derivative and the filtered output is only calculated for the last sample.
Is this right?

To be a real moving average filter for calculating the next output only the oldest sample should be removed from the input data and one new sample should be add, to have the 24 new input samples. 
But than the MPU FIFO has no big advantage if we have to transfer single values and have again a FIFO in the µC (unless we might use a hardware fifo on the µC too. I have not looked yet how the spi connection works, but a automated dma transfer to a fifo, if the spi peripheral has one, could reduce cpu load).

When one filter output is computed with 24 input samples, than everything is reset and started with complete new 24 samples it seems to be no real moving average filter to me (but i am no expert on filters).

I am also cant say from just a short look if we can run that at 1.333kHz (32kHz/24 Samples), as we have to transfer the 24 samples (we have to test how much time that needs), compute the filter (which has not as much computations as a large fir filter, but seems to be more than a small 1or 2 tab IIR filter, or a simple averaging filter) and need enough time for the rest of the system till the next 24 samples have to be read (24 * 2byte *3axis = 144bytes).

When advanced filtering is done, i think it should be not only done for MPU9250 but also for MPU6000 which has 8kHz max. internal sample rate, so i am also interested how a CIC filter would work with this.

At the moment i don't know where the bottleneck of the MPU data transfer and data processing is.
When i compare my 2nd & 3rd implementation with the same control rate, i can see that either the data transfer or the low level data processing&averaging in the MPU task must be the reason for the higher CPU load (everything that happens later/behind the MPU task is the same in both cases, shown be the same control rate of 2kHz)

MPU rate | IRQ sub-sampling  | Task sub-sampling | control rate | CPU load
------------ | ------------ | ------------- | ------------- | -------------
8 kHz | 1 | 4 | 2 kHz | 90%
8 kHz | 4 | 1 | 2 kHz | 54%
I can do some small variations in my implementations to do more tests to get more detailed results, but  i think maybe the data transfer is more responsible for the cpu load, then the data processing.

So i want to look how the spi data transfer is done, and if we can get some better performance there. Some possibilities could be higher clock (when it is not at max speed yet), larger data transfers from the mpu fifo (instead of "single" samples, i don't know yet how much overhead is there) or, when not used yet, using dma data transfers.

And one last note for now:
I don't know if we have drop the usage of the interrupt. Using the interrupt at high frequencies to just count the mpu new data  events, might have no large cpu usage (have to test this) and could be used to trigger the fifo data transfer from the MPU e.g when we have 24 new samples, which would than still be synchronized to the latest MPU data.

mlyle, 2015-07-20:
CIC decimation is the defacto technique for high ratio downsampling (>5) because it's linear in phase and cheap.  When the passband needs to be super flat it is sometimes cleaned up with a FIR as a last processing step.

Comparing a first order IIR, which is multiply (1 cycle), multiply-accumulate for floating point (3 cycles) = 4 cycles to a first-order CIC (add-- 1 cycle, plus one subtract at the decimated rate, so 1/24 cycle = 1.04 cycles) .. It's pretty clear that CIC is cheaper.

Trex4Git, 2015-07-20:
I thought in the above 3 stage example  there would be for 1 filtered and downsampled output value 24x3 Additions (integrate state), (24 additions (counter)) 1x3 subtraction (derivative state) and 1 division (output).
Your illustration above seems to implement the filter structure that that is shown in figure 9 A for a 1 tab CIC filter.
But the 1 tab CIC filter would still have 24x1 Additions (integrate state), (24 additions (counter)) 1x1 substraction (derivative state) and 1 division (output). 
A simple average has 24 additions and 1 division, has a little bit less impact, and a 1 tab iir has more cpu usage.

In the article it is also mentioned that for such a decimation usage of a CIC an additional FIR LPF on the outputs is often recommended.

Please dont't understand me wrong, i don't want to argue against some solutions or hype other. I just want to get a good understanding what solutions are the best and worth to implement.

And as said earlier i want to do step by step implementations and tests. For Example in my Task sub-sampling implementation it would be relative easy to change the averaging to a CIC filtering. Or implement the mpu fifo data acquisition without changing anything much after that.

mlyle, 2015-07-20:
OK, we can view it in terms of cycles per output, in which case you're right-- I had written it in cycles per input)  Of course, the counter can be basically avoided.

For some decimation uses where you need a very flat passband-- radio frequency, audio, etc, CIC is used and followed (or preceded) with a FIR chosen to flatten the passband a bit more.  Sometimes there's a last step of FIR downsampling.  But we don't really care about a small rolloff in response inside the passband.  That is, having a 600Hz response inside a 1.333KHz sampling rate which is -1dB at 300Hz and is -3 dB at 600Hz is better than our other choice of having a very steep filter around 256Hz.

More than any of these things, I care about the latency and delay.  We can introduce much less delay than the existing MPU-9250 filters by getting into the 32KSPS mode.

I'm OK with an incremental implementation as long as we can get to this kind of optimal implementation.  It kind of seems like 32KSPS, FIFO/wakeup management, SPI, and CIC/filter will have to be pretty tightly integrated in order to achieve the rate with minimal cpu use.

peabody124, 2015-07-21:
I'll defer the details of filtering to the experts.

One small comment - in the past with the FIFO I've been frustrated with the fact that you can get out of phase on the invensense parts and end up completely screwed. It's not a reason not to use them, but it leaves no room for error. I'd strongly recommend getting the code up and running and then do things like using SI to test if there is an improved tau -- I think that should be a pretty reliable objective metric. My guess, a 1ms improvement in group delay will be worth it. It should be fairly straightforward from what @Trex4Git has already done to test this.

From my experience you can also go a long way with a simple FIR boxcar, at least at first. Just accumulate samples over the window and then average them. You _do_ want to apply the same processing to the gyro and accel data though - just to make sure the complementary filter is seeing equivalently delayed results.

Trex4Git, 2015-07-21:
I understand that equal delayed gyro and accel are important, but when running at full speed , processing them the same might be difficult, as there are already differences in the MPU sampling (gyro max. 32kHz, accel max. 4 kHz in MPU9250, both mode not yet supported in taulabs, and 8kHz/1kHz for MPU6000). 
So when the delay data in the invense datasheed are reliable will still have to take care of our filtering and data acquisition (gyro fifo, accel & temperature normal?).

Even at the moment when both gyro und accel are setup with DLPF settings for 184Hz gyro should have a delay of 2.9ms and accel a delay of 5.8ms. Is this difference in delay already somewhere compensated?

peabody124, 2015-07-21:
Good point. I'd forgotten they couldn't be made entirely equal. Just make sure not to drop any fresh accel samples. They should all be consumed if new and filtered somehow.

Trex4Git, 2015-07-21:
I want to do some  first implementations for testing, and see if its worth to do more development on the different topics.

For example:
* Implementing a new filter should not be a big task, and if other with more experience can test it also it would be nice
 * I am also very interested how a CIC filter with downsampling functionality performs. If its output can be used without disadvantages for the integration in the attidutde module and if the IIR LPF in stabilization module may also not be needed anymore.

* I have already seen that we can speedup the SPI clock from 2,625MHz (what we currently use) to 21MHz (8x faster) (For the configuration it must run at a slower speed, as only the interrupt and data register can be read at high speeds, but this is already been done in taulabs, with 100kHz SPI clock during configuration of the MPU). Some test how this may have advantages on the CPU load are easy to do for me.

* I am still not sure about implementing the FIFO, and it has not the highest priority for me. As you have reported about problems i did some further research and found also more reporting about problems with the FIFO. So when it will be implemented a lot of testing and bugfixes/workarounds for problems might be necessary
 * I have seen that OP had earlier used the FIFO to get the data from the MPU, but at the end of last year at first there was a alternative direct reading from the MPU added, and later the FIFO reading was more or less completely removed. I don't know if there where also problems with the FIFO, (there was a FIFO reset functionality implemented). 
 * What they now use seems to be similar to what i have already implemented here or we are planing for further implementations. From my understanding they run the MPU SPI Interface at 21MHz, no DLPF is used and the MPU sampling rate is 8kHz (for revo & revo nano), than there is a simple averaging down to 500Hz, and the rest of the control system is run with that rate. In the inner loop they also have the IIR LPF (which is also newly implemented in cleanflight by BorisB and there are reports about a much better performance with this).

Trex4Git, 2015-07-21:
Hi Alessio, thanks for your answer and the information. So this is the first cooperation between LibrePilot and Taulabs  on development ? ;) 
When i searched for FIFO usage with MPU6000 and problems I found also some links to OP, so i had a look at it. Than i have seen also some similarities between your current  implementation and what I have done or plan to do.

After that i have seen that the FIFO usage seems to be from days before Taulabs split, so Taulabs had also FIFO functionality but removed it a bit earlier (2013: https://github.com/TauLabs/TauLabs/commit/c950a55dbba6e561acf0bc7cea77fa3f3580440e)

Did you also do some performance tests regarding the CPU load, it would be interesting, at least for me.

As it was only a small change, i did some tests with the changed SPI clock from 2,625MHz to 21MHz:

MPU rate | IRQ sub-sampling (Drop)  | Task sub-sampling (LPF) | control rate | CPU load 21MHz  |  CPU load 2.625MHz
------------ | ------------ | ------------- | ------------- | ------------- | -------------
8 kHz | 2 | 4 | 1 kHz | 32% | 48%
8 kHz | 4 | 1 | 2 kHz | 46% | 54%
8 kHz | 1 | 4 | 2 kHz | 59% | 90%
8 kHz | 1 | 8 | 1 kHz | 40% | (72%)
8 kHz | 1 | 16 | 500 Hz | 31% | 
8 kHz | 1 | 32 | 250 Hz | 26% | 
This has already a nice improvement in CPU load. 

I am not sure if a PIO data transfer from a FIFO with all input samples for 1 output samples has an big advantage. 
At the transfer itself i see no overhead if we transfer e.g. 8 times 6byte for one gyro sample (3axis x 2 bytes) or
1 time 46 bytes for 8 gyro samples from the FIFO: https://github.com/TauLabs/TauLabs/blob/next/flight/PiOS/STM32F4xx/pios_spi.c#L583-L602 
And at the rest of the function i see at a quick look that there seems to be only some very small overhead with some security checks.

The FIFO transfer might has an advantage if we use the 32kHz mode on MPU9250 and collect more samples. Than maybe the DMA transfer instead of PIO has an advantage (for a small amount of samples the overhead of DMA Block Transfer seems much larger then for the PIO Block Transfer).

With my current implementation at each data transfer not only gyro data is transferred, but also temperature and accel data. Only further processing on temp and accel is not done.
Accel data are sampled in the MPU6000 with max. 1kHz, and for the temperature i have not seen data of the sample rate.
So a further performance improvement might be achieved, when temp and accel data are not transferred at the high frequency of gyro sampling, but only at the last gyro sample.
.
.
.
.
Maybe i am writing here to much, and change that in future, but at the moment its nice also as a documentation for myself.

Trex4Git, 2015-07-21:
Quick implementation and test with accel and temperature data transfer at gyr sub-sampled frequency:

MPU rate | IRQ sub-sampling  | Task sub-sampling | control rate | CPU load
------------ | ------------ | ------------- | ------------- | -------------
8 kHz | 1 | 4 | 2 kHz | 55%
8 kHz | 1 | 8 | 1 kHz | 36%
8 kHz | 1 | 16 | 500 Hz | 26%

Again some improvements.

Trex4Git, 2015-07-23:
Implemented an enhancement for MPU6000 accel & temperature sampling
- sample accel and temperature data at. min. 1kHz, when gyro sample freq. is >= 1kHz
- if lower gyro sub-sampling/output freq. then 1kHz is used:
	- simple averaging of accel and temperature data is done

MPU rate | IRQ sub-sampling (Drop)  | Task sub-sampling (LPF) | control rate | CPU load
------------ | ------------ | ------------- | ------------- | -------------
8 kHz | 1 | 4 | 2 kHz | 55%
8 kHz | 1 | 8 | 1 kHz | 36%
8 kHz | 1 | 16 | 500 Hz | 29%

Performance as expected.  A bit more CPU load when control rate is 500Hz, as than accel & temperature data are sampled/transferred twice and averaged for 1 new output value of the MPU driver.

Trex4Git, 2015-07-23:
And i am already working on the CIC implementation and did some research on CIC filters.

@mlyle : How I understand the CIC filter I think there is some bug in your code sketch.
I understand that it should be:
```
uint32_t integrateState[3]={};
uint32_t derivativeState[3]={};
uint32_t sampleNum=0;

void HandleInput(uint16_t gyro) {
    integrateState[0] += gyro;
    integrateState[1] += integrateState[0];
    integrateState[2] += integrateState[1];

    sampleNum++;  if (sampleNum < 24) return;  // discard 23/24 samples.
    sampleNum = 0;

    derivativeState[0] =  integrateState[2] - integrateState_old[2];
    derivativeState[1] = derivativeState[0] - derivativeState_old[0];
    derivativeState[2] = derivativeState[1] - derivativeState_old[1];

    integrateState_old = integrateState[2] ;
    derivativeState_old[0] = derivativeState[0] ;
    derivativeState_old[1] = derivativeState[1] ;

    float output = derivativeState[2] / 13824.0f;      // 24 * 24 * 24
    Output(output);
} 
```

mlyle, 2015-07-23:
Yes, sorry, that's correct, minus subscript on integrateState_old[].

Trex4Git, 2015-07-23:
Yeah, that's a typo writing it to fast. The code i prepare for implementation looks a little bit different, so i just copied your code, and edit it, to show my understating of the CIC.

I plan to implement for the testing #defines so that you at least change before building the firmware what to use (I thought about CIC 1 tab, CIC 2, CIC 3, simple averaging). But want to look at fist on the behavior of the filter with different settings, to see what should be implemented. 

So what cut-off frequency do we want to have? And what is the min. passband width?
Maybe we also a compensation filters on the output of the CIC.

mlyle, 2015-07-23:
Usually we don't talk about cutoff frequency with CIC because of the sinc response.  The shape pretty much comes from the downsampling and the order of the filter.

I would think we would want a second or third order filter.  At 8KHz we could downsample /8 for a 1khz flight loop that would respond well out to 400hz. (But the 256hz prefiltering would limit this).

Trex4Git, 2015-07-24:
As already said l Iooked  on the behavior of the filter with different settings, to see what should be implemented. 
Here are some results:

At first some magnitude responses for different setups in a wideband plot:

[image]

That it gets to 0dB at multiples of the the MPU rate seems to be a bit strange. Maybe the formula doesnt work for this frequency range. When everything is correct this might give some problems with aliasing.

I don't know if everything is correct as i have not worked with filters since university. I looked at http://www.embedded.com/design/configurable-systems/4006446/Understanding-cascaded-integrator-comb-filters and some other websites and papers. But sometimes there are also some other descriptions.
But i compared my results to the  above mentioned site, and i seems to give the same results. 

Here is a zoom of the above plot to the nyquist frequency range for 2kHz output:

[image]

And again the same frequency range, but some other CIC setups:

[image]

There are already some parameters to tune a CIC filter, and it depends what behavior is desired.
Especially if there are experiences how large the passband shout be, it would be nice to get some.
Than i can look what setup performs best.

For example a 2 stage CIC with N=2, seems to perform better than a 3 stage with N=1 (both on 8kHz MPU gyro sampling, and 2kHz output frequency; the green curve and the dashed blue curve in the second plot).

When that might get merged in the future I thing it would be nice to have 1 or max. 2 parameters where users can tune the behavior, but it should not be too complex.
But we have to see if such a setup is possible. At to moment the 1tab IIR  gyro LPF in the stabilization task, has also no constant behavior, but can also change even when GyroTau is not changed. Because sees pandpass bandwith seems to depend on the linear on the MPU rate. The frequency were the output is 3dB lower then at 0Hz seems to be (with the default GyroTau of 0.005) 0.1625 * f_nyquist = 0.1625 * f_gyro / 2
(at 500Hz MPU Sampling ~40.6Hz and at 2kHz MPU rate 162,5Hz). So lower than i thought at first when i did the plots above.

So experience with  that an optimal settings for GyroTau are very interesting for me.

In literature is often a compensation filter at the output of the CIC recommended. So i did some tests with a 2tab IIR filter (when this works good, i thing the 1tab IIR in stabilization can maybe be removed). It seems to be possible to correct the unwanted decay in the passband, and get a steeper decay to the stopband. Especially here i am not sure if everything is done right, as its was a very quick coding and generating the data.

[image]

The first sub-plot shows the magnitude response of 2tab (green) and a 3tab (blue) CIC with otherwise the same settings.
In the second plot are the magnitude responses of 2tab IIR LPF to combine with the CIC. The color shows for which CIC filters they are used.
And in the third plot is the combination of the CIC and the different LPF shown. I have done that by a multiplication of magnitude responses of the CIC filter with the IIR LPF. But I am not shure if this is right way to do it.

mlyle, 2015-07-24:
I can't read this in detail now.   But the repeating is because 8khz is already aliased to 0 hz on the input at fs=8khz.   That is, the filter plot is only meaningful from 0 to 4khz at fs=8khz (nyquist) for real samples (-4 to 4khz for complex samples)

mlyle, 2015-07-25:
So, ... I don't think frequency response is the most important thing here actually.  In my opinion there's three important things:

First, it should be really flat up to 100Hz or so, because these are the actual dynamic frequencies the system can respond meaningfully at.  Any additional response at frequencies below the Nyquist frequency is a bonus, but not the most important thing-- especially as we probably have less phase margin up there.

Second, it should have a small response to aliases-- especially aliases at lower frequencies and 0Hz because these would cause a biasing of the attitude solution.  I think all of the filters achieve this.

Third, it should have the smallest possible delay, because this improves control system bandwidth and perceived responsiveness.  Also, large delays will cost phase margin and possibly upset system stability.

So, basically, I'm not sure how much it is worth trying to flatten response of the system above 200Hz, especially if it adds further delay.

Trex4Git, 2015-07-25:
Thanks for your feedback and the informations. As already said i have in principal some knowledge about signal processing and filtering, but  its some time ago, and was almost only theoretical stuff.

One point that i also find interesting is, how this might behave on different targets. 
- My only FC is the Quanton with the MPU6000. 
-Sparky2 has the same µC but uses MPU9250 which can have higher sampling rates in the MPU (so when this implementation is almost finished i want to Port it to Sparky2 and MPU9250, and we can test if there is an advantage with these higher sampling rates). 
- The CC3D uses also the MPU6000 like the Quanton but has the F1 instead of F4 with much less processing power. It would be interesting if the higher SPI speed will also work without problems, and what might be the performance advantage of it.
 - I think at the moment the CC3D could not run well with 1kHz MPU rate, but it would be nice if it would be able to do that with these changes.
 - And it is interesting if the 500Hz control rate could be used  in combination with the new filtering stuff.

So it would be good, if the new filtering could be easy adapted to different targets or different setups. 
Some people might prefer a fast acro setup with no additional modules. Other might prefer to use a lot of modules and functionality and want to have less CPU load from the lower control system.

In the following plot the solid lines show the frequency response of the 1tab IIR gyro LPF in stabilization for different MPU rates (Fs). When you normalize the plot to the sampling frequency, the curve looks the same for the setups, but when plotted versus the physical frequency you can see the differences.
With higher sampling rates you get a filter with a expanded passband.

[image]

I don't know if the improvements of higher sample rates some people have reported (and in the autotune thread you can see e.g lower tau with higher rates) are just because of the faster control loop, or if the change of the gyro filter has also a significant influence.

I am also not sure if that was really the intention of the implementation, or if using higher rates was not in focus when it was done.

So for a quick theoretic test i changed the algorithm a little bit, an instead of using a constant fakeDt, it is scaled with the real Dt and normalized to a behavior like the 1kHz MPU rate has with the default settings of GyroTau. 
As you can see at the dotted lines:
At least for low frequency this quick scaling test shows similar behavior in frequency response for the different sampling rates.

In principal the same behavior can be seen for CIC filter compare different output frequencies but the same filter order (M) and the same differential delay (N), what results in the same algorithm that could be used for different output rates
In the next plot the solid lines show again as reference the frequency response of the 1tab IIR gyro LPF in stabilization.
The color encodes the output frequency, and the line style the type of the filter. The dotted lines show a 3tab CIC with a differential delay of 1, and the dash-dotted lines show a 1tab CIC with a differential delay of 2. These are just some examples that might be used later, but with changing the parameters you can design the behavior of the filter (i think i got now a feeling how it can be tuned).

[image]

You can also adjust the CIC filter to get a very similar frequency response for different output sample rates (at least for some, but not for all cases possible i think).

It can be also seen that the CIC, even without a additional compensation filter, can be tuned to a certain extend.
The shown examples already have a flatter response up to 100Hz compared to the default settings of the current gyro LPF.

mlyle, 2015-07-25:
I think the main improvement of higher gyro filter frequencies is reduced delay through the gyro filter.  Any filter delay shows directly in system open loop response.

mlyle, 2015-07-25:
The main thing I see in these plots is that the frequency response of the first order CIC is *almost* OK.  The second nyquist zone is down -13dB.  So we might not need third order and could maybe use second order.

Trex4Git, 2015-07-25:
Thats true, the plots above should only be examples how the behavior can be changed with the parameters, not final filter setups.

For applying the Nyquist rule i think we have to use at the output of the CIC filter also the output sampling frequency, not the input sampling frequency.
So the -13dB Peaks of the M2/N2 CIC filters are already above Fs,out/2. Also you can see that the M3/N1 filter are also only at around ~-11.5dB at Fs,out/2.

So in figure 1 & 2 (i edited the posts above to give the figures numbers) there is already a comparison for M3/N1, M3/N2 and M2N/2 filter. But i also changed figure 6, and added M2/N2 as a additional curve. There the first peaks above Fs,out/2 are reduced to ~-26dB

To have a better filtering close to Fs,out/2 and above we have to increase the order of the filter (M) or the differential delay (N) further. But than also the response <200Hz is not so flat anymore, and we get an larger decay here also.

To correct this, we would need a additional compensation filter, when we have have decided what behavior is still acceptable.

mlyle, 2015-07-25:
I'm not sure I understand. For instance, I'm looking at the dashed green plot.  Fs=500hz,  Nyquist =250 Hz.  All of the products to the right of 250hz are down at least 13dB (peaking at 360hz which will fold down to 140hz).  which is pretty good.  

I still think we should use more than a first order filter, but the fact 2nd order would be down 26db tells me second or third order is definitely good enough.

Trex4Git, 2015-07-25:
That is what i wanted to mention.
I just thought that the -13dB peak that is folded back for the M1/N2 CIC is too high (M3/N1 looks also not so good with -11.5dB at the Nyquist frequency, and not so big decay). 
If -26dB are enough and the decay of the M2/N2 CIC <200Hz is acceptable i think this might be a good compromise.
As the M3/N2 has better ~-40dB at the first peak, but even more decay <200Hz in the magnitude response.

Mx/Ny means filter has the order x, and a differential delay of  y.

mlyle, 2015-07-25:
OK.  I am having a lot of trouble reading the plots.  

I don't think we need to consider different R; the response is relative to R as long as R is relatively large (>=5).  

Can you plot like, an output frequency of 1KHz, M=2, N=1 easily, for the first couple nyquist zones only.. and on a scale to -30dB?  Ideally the scale would show where -3dB is too.

Trex4Git, 2015-07-26:
Maybe there is too much data in the plots.
I just plotted different R (16,8,4) to get a feeling if its really more or less independent of R (when related to each output frequency). And I also wanted to see the behavior of different output frequency not relative to each output frequency, but relative to the absolute (not normalized) frequency. 
Because when we disable DLPF, in MPU6000 datasheet it is said  that there is 256Hz bandwidth, and I think it would be nice if we can make use of it, and don't cut to much away because of the CIC decay in the passband.

As the plots are generated automatic with scripts, its easy to do just change some parameters and generate a new plot. 

[image]

I think  we should choose N=2, as the zeros are located  F<sub>s,in</sub> / R / N = F<sub>s,out</sub> / N. 
With N=1 the first zero is at F<sub>s,out</sub> = 2 * F<sub>nyquist</sub>, with N=2 the first zero is at F<sub>nyquist</sub>. Than we can adjust with M how much the peaks at frequencies > F<sub>nyquist</sub> (which fold back to the passband) should be damped.
But a larger damping in the stopband is combined also with a larger decay in the passband.

Trex4Git, 2015-07-26:
For comparison: Filter order M = 1; differential delay N =2 other settings as above

[image]

mlyle, 2015-07-26:
OK, I understand.

I see why you prefer the filters with differential delay of 2.

Well, maybe we should use M=2, N=2.  We can always add some pre or post-emphasis with an IIR later if we want frequency response over 180Hz (the -3dB point if we double the attenuation above). 

Actually, I kind of wonder if the 184Hz/92Hz/etc lowpass are CIC on the MPU-9250.  They look like they follow the exact pattern in delay and bandwidth that the above would dictate, and I wouldn't be surprised if our M=2, N=2 filter had a -3dB bandwidth of 184Hz.  :P

Trex4Git, 2015-07-26:
Do you have some detailed information about the filters in the MPU ? I could not find any description, other then the register description table for the DLPF settings.

I updated my plot script so here again a new version for the 2 plots above, and also M3/N1, M2/N2,  M3/N2.
- The red region are frequencies above F<sub>nyquist</sub>/2, signal there will fold back to lower frequencies because of aliasing.
 - So I show also in the dashed lines these folded frequencies with a origin up to 4kHz (because higher frequencies should be cut off by a anti-aliasing filter in the MPU).
- The green region is 0-256Hz. There should be usable information when the DLPF in the MPU is disabled.
- The white region shows a range where signals in there would be processed by our control loops, but that should be only noise, as there is no data from the MPU in this region, and it could be cut away with a additional filter.

figure 6_1: M=2, N=1
[image]

figure 7_1: M=1, N=2
[image]

figure 8: M=2, N=2
[image]
The -3dB point is here ~160Hz, so maybe the MPU internal filtering has for control rates of 1kHz or lower no disadvantage, or an advantage (184Hz bandwidth on lowest DLPF setting).

figure 9: M=3, N=2
[image]

figure 10: M=3, N=1
[image]

So when an additional filter is used, i think the M2/N1 or M3/N1 can be a better base than the N=2 filter.

Because they have lower decay in the green region (less work for the filter to flatten that), and back folded signals in the white region can be cut off with the additional filter.

mlyle, 2015-07-26:
Very nice plots.  It makes it easy to understand the tradeoffs.

There's one other factor though.  This all assumes the MPU-9250 is flat up to 250Hz, when its own bandwidth is probably actually specified as a -3dB point.

This is starting to look less compelling for the 8KHz to 1KHz downsampling.  There's a small advantage in total delay (1ms) with the N=1 filters.

Another compromise would be to downsample /7 or /6 instead.  This would increase the flight loop frequency, move the second nyquist zone higher in frequency (which makes the filter more effective), flatten lower frequency response and remove delay.

Trex4Git, 2015-07-26:
When we use a additional filter and assume that the MPU has at 250Hz already a 3dB decay, we may can design the filter also to compensate this.

So in the comment above after figure 8 i also already said when running at a output frequency of 1kHz i am not sure if using our own filtering would have an advantage. When using 1kHz sample rate in the MPU and a DLPF of 1 MPU6000 should have 188Hz bandwidth and 1.9ms delay, and MPU9250 184Hz bandwidth and 2.9ms delay.
So maybe its not worth to use our own solution for that.

But my main interest for working on this stuff was anyway higher control rate then 1kHz, as that is not really usable with the current implementation on the Sparky2 with the MPU9250.
For my Quanton with MPU6000 you can already set a sample rate divider based on 8kHz clock in the MPU, when DLPF is disabled. But than possible no filtering in the MPU is done.

So the things I am working on, should primary have advantages for control rates above 1kHz on targets/setups that could handle it (the update of SPI clock should bring also improvements on the CPU load with the CC3D I think).

So i primary showed here the 8 times downsampling variant, because you requested it. 
I also looked on higher control rates, and there the situation looks a little bit better, with maybe less need for a additional filter.

My target would be 2kHz or 2.666kHz, but we have to see, how the CPU load would be with this.

figure 11: M=2, N=1 @2kHz output
[image]

And when a additional filter is used, i want to try it with a second order IIR, where 2 of the 5 coefficients are 1, so it should be 3 multiplications and 4 additions that have have be done at the speed of the output frequency.
I don`t if there is an advantage in using the CMSIS DSP functionality or not, especially if it would recognize that 2 coefficients are equal to 1.
http://www.keil.com/pack/doc/CMSIS/DSP/html/group___biquad_cascade_d_f2_t.html

mlyle, 2015-07-26:
We can't compensate for the MPU-9250's filter unless we know its shape.  The most we can do is hope to flatten our own response, at the cost of more delay.

Trex4Git, 2015-07-27:
I did now also look on the delay of the filter.
For a CIC filter the group delay should be defined:
- In samples: sample_delay = M * (D - 1)/2 = M * (N*R - 1)/2 
- Or expressed in time (based input sample rate): delay = M * (D - 1)/2  * (1 / Fs,in) = M * (N*R - 1)/2 * (1 / Fs,in)
- R=8, F<sub>s,out</sub>= 1kHz
 - M=2, N=1: 7 samples / 0.875ms
 - M=3, N=1: 10.5 samples / 1.3125ms
 - M=2, N=2: 15 samples, 1.875ms
- R=4, F<sub>s,out</sub>= 2kHz
 - M=2, N=1: 3 samples / 0.375ms
- The DLPF,when set to 1 and 188/184 bandwidth, has for MPU6000/9250 a delay of 0.92ms/1.93ms (0.98/0.97ms with DLPF disbaled vs:1.9/2.9ms with DLPF=1)

- The current 1tab IIR LPF has with

Trex4Git, 2015-07-28:
As additional comparison for 2kHz output frequency:

figure 13: compensation for R=4, M=2, N=2  @ 2kHz output
[image]

- CIC filter R=4, F<sub>s,out</sub>= 2kHz
 - M=2, N=2: 7 samples / 0.875ms
- compensation filter for this setup (not as necessary as for M=2, N=1)
  - a delay of  0.91 output samples (0.455ms) @ 0Hz; together with the CIC 1.33ms
  - a delay of  1.08 output samples (0.54ms) @ 100Hz; together with the CIC 1.415ms
  - a delay of  2.1 output samples (1.05ms) @ 250Hz; together with the CIC 1.925ms

I think this could also be used without a additional compensation filter, as having less aliasing, and quicker decay in the stoppband than the "R=4,M=2,N=1" filter.
Also as only being the CIC filter it has a constant delay regarding to the frequency, but the "R=4,M=2,N=1"+its additional compensations filter have for lower frequencies even slighly less delay, than the N=2 CIC alone.
So the N=2 CIC alone has less aliasing, but more decay in the passband (tests have to show if that is a advantage or disadvantage) compared to the N=1+compensation.

mlyle, 2015-07-28:
So, one thing that's of note is that lots of ESCs seem to have trouble receiving oneshot data faster than 900Hz.

One option is we could downsample /5, for 1600Hz output, run the flight loop updating attitude at 1600Hz, and generate our output at /2 = 800Hz for the ESCs.

Trex4Git, 2015-07-29:
This is interesting. 
But at first i want to get the principal things done, do some test, port to MPU9250 & Sparky 2, maybe also to CC3D, so that other can do also some test.
When than maybe we have some results and found good settings, I wanted to think about maybe adding additional functionality, or preparing a PR before doing more fancy stuff ;).

Regarding the ESC i don't know how SimonK firmware deals with it, but in BlHeli it should work since V14, as I suggested it to sskaug and he implemented & tested it up to 4kHz (https://github.com/bitdump/BLHeli/issues/55#issuecomment-99584980), what is the physical limit of a Oneshot125 signal (125-250µs pulse width).

fujin reported yesterday that it does not work with BrainFPV and 1kHz. So it would be nice if he can take a look at the output signal with a scope or logic analyzer. 
When i had build my first (and only up to now) quad, had no experience and did some first test. I had not so constant Oneshot125 output from my Quanton (you can see that in the link i posted above).

But since than also the OneShot (SnycPWM + HighRes) stuff in Taulabs has improved and when i looked with a logic analyzer at the output signal after my changes here, i had not seen such unwanted behavior in the output signal.
But that was after the performance improvements, I think. When  I remember correctly before, especially at high CPU loads, it looked even worse sometimes.

And do you think it makes sense to run the control loop with a higher rate, then output signals are generated? The system can not react to every second control signal.
Otherwise we could just downsample by 9 or 10 to get 889Hz or 800Hz control rate.

Do you have experiences what the gyro signal on a quad looks like, especially where are useful information, and where is only noise?
I have written my questions and concerns to the forums, as it is related to this issue, but not directly the focus of it: http://forum.taulabs.org/viewtopic.php?f=17&t=305#p6473
There i posted also a link to a Cleanflight development thread. The measurements there show that useful information seems to be even below 100Hz, and above only noise. But maybe they have more aliasing problems, then we might have with the CIC downsampling.

mlyle, 2015-07-29:
> implemented & tested it up to 4kHz (bitdump/BLHeli#55 (comment)

So, the "useful information" depends on a number of different transfer
functions that are multiplied.  Basically, there is mechanical delay,
ESC/motor delay, and filter delays and frequency responses that combine to
make an overall system transfer function.  Removing delay from any of these
steps will create higher system bandwidth, too.  So right now the useful
information is from 100Hz and down, but removing a couple of milliseconds
of delay from the input will possibly increase that frequency a little.
Likewise, generating motor output more frequently will shave a fraction of
a millisecond off.

This is why I've been saying that there's only two things that really
matter-- reducing delay and preventing aliasing.  Reducing delay allows you
to run higher control gains without oscillation, and that makes pilots
happier.

As to the "react to every other sample"-- yes, we can, in the attitude
modes where we can integrate our attitude more precisely.  Integrating
small steps works better than integrating large steps, and seeing the phase
difference of yaw starting a little bit before roll instead of at the same
time could result in a fairly big attitude measurement difference.

One thing we have noticed experimentally is that there's a number of
situations where our attitude solution can get biased.  Some of these
relate to high G-forces where the accelerometer induces bias.  But it can
also happen from unbalanced props.  This could be the effect of aliasing or
nonlinearities creating mixer products, and could very possibly be much
better with a higher sampling rate (which lowers the amplitude of the
aliases and moves them to higher frequencies, all else being equal).

Ultimately, I'd like to figure out how to get torque / power measurements
from smart ESCs.  This stuff obviously changes faster than the quad's
rotation shows up in the gyros.  That would let us run a very tight
power-imbalance (angular acceleration) control loop inside the rate loop.
But since that requires hardware that isn't commonly available, it'd be
nice to get the rate loop delay as low as possible and as good of input as
possible to the attitude loop.

Trex4Git, 2015-07-29:
The lowest useful delay with the CIC filter seems to be M=2/N=1 for high output rates (~2kHz), and M=3/N=1 for lower output rates (~1kHz). I don't know if aliasing is already damped enough, or if a additional filter is needed.
So when you combine these with a additional filter, like i have done e.g. with the above shown compensation filter, you can flatten your passband (or also damp it more if needed), and put higher damping on the (aliased) noise.
The delay for f < 100Hz can than still be smaller than just using a M=2/N=2 CIC filter without a additional filter. As also not being constant, the group delay <100Hz only has small variations

But in the future the M=2/N=2 might be a good default for all output frequencies & setups as providing a acceptable anti-aliasing (with the first zero at f<sub>nyquist</sub>), a constant (group) delay (that is not much larger than the N=1 CIC filter+compensation filter), and maybe the decay in the passband is also an advantage.

In setups with very low noise (vibrations,...) the N=1 type CIC filters might also have a good performance when used without a additional filter, and you will have with the lowest delay.
The usage of a additional/compensation filter should than only be  for experts, as the behavior of the 2tab IIR in combination with the CIC is not as easy to tune as a simple 1tab LPF.

But I am still concerned about how much noise is in the gyro signal. Even when we have with a reduced delay  "useful information" at higher frequencies, if there is also noise at these frequencies  I am not sure if it would be better to filter both, signal & noise.

Next steps I plan to do is implementing a configurable CIC filter, maybe a additional compensation filter, and switches to enable/disable all the filter (also the 1tab IIR in stabilization).

After that, testing must be done. But I still have no experience with the logging functionality in Taulabs, and from comments on IRC it seems also to be complicated.