TauLabs/TauLabs #1736: PFD only displays heading from internal mag

lolorc, 2015-07-21:
Hi there,

I'm playing with i2c at the moment, I've just plug a HMC5883 (GY-273) to do some tests with my quanton.
I had to erase all settings to have it working, that's not the point of this issue. (uart3 set to i2c, mag set to external on uart3)

While the mag scope uses data from the external mag, the PFD gadget doesn't.
If I rotate the external mag the mag scope shows the changes but the PFD heading doesn't change.
if I rotate the FC the scope stays still and the PFD compass follows the FC rotation.

peabody124, 2015-07-21:
It is because the sensor fusion mostly looks at the gyros for rapid change. Change the mag and wait a minute.

lolorc, 2015-07-21:
Oh ok thanks for the answer.
I'm closing this non-issue then.
the compass on the PFD is always slowly moving, it does 360° in about 80seconds, can't really do the test

closed 2015-07-21 by lolorc