TauLabs/TauLabs #1747: Can not update firmware to 20150713-4b9fc7dc46 on Quanton

JeromePilot, 2015-07-24:
Hi,

My quanton is on firmware 20150131-4082e9d8. I have flashed it easaly under the GCS under XP several times without pb. 

After update with "Windows (installer)" to 20150713-4b9fc7dc46 the quanton is seen (it says me the firmware is the matching 20150713) but not possible fo flash the new firmware, got a red error message. 
I tried to flash using the 20150131-4082e9d8 GCS : no problem to flash 20150131-4082e9d8 firmware but impossible to flash 20150131-4082e9d8 neither under the previous GCS.

I suspect the firmware for quanton under 20150713-4b9fc7dc46 is not correct.

Thk,

Jerome

guilhermito, 2015-07-24:
You need an updated bootloader, the flash size available for fw was changed

closed 2015-07-24 by guilhermito

JeromePilot, 2015-07-25:
Thk for your prompt reply ;o)

 

Do you have a link with the appropriate bootloader ? I cant find one even
if I follow the wiki
(https://github.com/TauLabs/TauLabs/wiki/Flashing-bootloaders)

 

Cheers and thk again

 

Jerome

 

  _____  

Quanton (#1747)

 

<https://github.com/TauLabs/TauLabs/issues/1747#issuecomment-124763574>  it
on GitHub.
<https://github.com/notifications/beacon/AM3Vn6a5CBa4Zvxg1sgErT4AuGAkFrxnks5
ogr_4gaJpZM4FfeQ3.gif> 

---
Ce courrier électronique ne contient aucun virus ou logiciel malveillant parce que la protection avast! Antivirus est active.
https://www.avast.com/antivirus

Trex4Git, 2015-07-25:
I updated my bootloader on the Quanton from the GCS.

I think it was the "bu_quanton.tlfw", when I remember correctly

JeromePilot, 2015-07-25:
Thk for your reply.

 

Update the bootloader by GCS is not working on my computer (XP may be). As
per the wiki (https://github.com/TauLabs/TauLabs/wiki/Flashing-bootloaders)
the bootloader update is by
<http://www.st.com/web/en/catalog/tools/PF258168> STM32 ST-LINK utility.

 

The files with extension .tlfw seems to be firmware. The bootloader have
extension .bin I think. If you have the latest bu_quanton.bin so I can
try.

 

Thk U

 

Jerome

 

  _____  

Quanton (#1747)

 

<https://github.com/notifications/beacon/AM3Vn5Gx5-7H-Kpb5Ndfs5ia4xx3k2jLks5
og6YEgaJpZM4FfeQ3.gif> 

---
https://www.avast.com/antivirus

JeromePilot, 2015-07-25:
Just check, the present bootloader on my board is V81.
[image]

JeromePilot, 2015-07-27:
Where can I find a newer bootloader than the V81 that is flashed on my
board. I have no linux machine where I could make it

 

Cheers

 

Jerome

 

  _____  

 

<https://github.com/notifications/beacon/AM3Vn6a5CBa4Zvxg1sgErT4AuGAkFrxnks5

https://www.avast.com/antivirus