TauLabs/TauLabs #1753: Pin out for CC3D PPM

MrSatoV, 2015-07-28:
Hey guys, I'm pulling my hair out over here.  I feel like I'm missing something super simple, but so far nobody has the answer.  
I see in this post that CC3D is working with PPM:
tracernz commented 8 days ago
Since fixing the stack sizes (5b98d5a) I have test flown this on both Naze32 rev. 5 and CC3D. Flew with a PPM RX in acro and horizon mode with full throttle punchouts etc. on a 4S powered 250 without issue. I'm happy with this now. :+1: I have also now reviewed all of the code and am satisfied. There is plenty of work to do before the target is fully supported but this PR will get us flyable.

But no matter what pin I use, I get no TX input.  I've seen some workarounds, like using sbus, but it sounds like tracernz isn't doing any kind of workaround, and just plain using PPM through the recvr port.  I am aware of the timer issue with using pin3, but that's easy enough to fix, and all other FC firmwares I've tried have an alternate pin assigned to use a different timer.  In OP it's 4 or 8 (8 for OneShot, which is what I want, and use with OP).  Of course I tried both, but neither give any results.  

What pin should I be using?  I tried them all, but at least if I know I have the right pin I can move on in the troubleshooting process.

tracernz, 2015-07-28:
I was using pin 3 with outputs set to 1 MHz (1000-2000 us pulse width), not "OneShot". I think using PPM with 12 MHz outputs on CC3D is broken, see #1754 .

MrSatoV, 2015-07-28:
Ahhh.  Thanks.  Sounds like it might get fixed in the next release.

peabody124, 2015-07-28:
See #1754 for relevant bug I made based on this.

Of course try this with the props off! But I think you should still be able to use the sync part of one shot (which is the important bit). Set them to OneShot but then change from 12MHz to 1MHz in the output configuration. You'll only have 125 "levels" of resolution but TBH I'm pretty confident you won't be able to tell the difference.

MrSatoV, 2015-07-28:
As soon as I get done with helping my daughter with my homework I'll give it a whirl and get back to you!

mlyle, 2015-08-05:
@MrSatoV did requesting 1MHz make it work?

soend, 2015-08-07:
Tried this with D4R-II connected to pin 3 in receiver port. Changed resolution in outputs to 1MHz but it didnt fix it.

mlyle, 2015-08-14:
Fixed by PR #1775.  It is now possible to use pin8 for PPM on CC3D, and this is now the default in GCS.

closed 2015-08-14 by mlyle