TauLabs/TauLabs #1754: PPM input should account for timer speed

peabody124, 2015-07-28:
As I understand it, PPM (and PWM) input when using 12MHz mode is currently broken. One workaround is to reassign the pin to a timer that is left at 1MHz. However, it shouldn't be too hard to have the pios_pwm and pios_ppm code to check the value in the prescalar to determine the period and account for this (including in the overflow event handler).

MrSatoV, 2015-07-28:
I have seen a suggestion to fix this elsewhere.  (http://forum.taulabs.org/viewtopic.php?f=11&t=769)
His solution doesn't include any checks, but it does assign it to a pin on a different timer.

mlyle, 2015-08-07:
The different timer is the previous issue you opened, #1753.

@peabody124 I haven't read the code, but is it just 12MHz that's the problem, or is it also SyncPWM?  Do we disable and re-enable the timer for oneshot?

peabody124, 2015-08-07:
Oh wait, I forgot that syncpwm actually halts the timer and resets it so I think they are actually incompatible

mlyle, 2015-08-08:
So this is much less of a win if it doesn't work with synchronous modes.  At the very least I'm removing from this release; maybe consider closing.

mlyle, 2015-08-12:
Closing:  discussed with @peabody124 .

1. This mostly only affects CC and we have a good workaround
2. Most of the time we would use 12MHz we'd also be using SyncPWM, which as implemented is fundamentally incompatible as currently implemented with PPM/input capture (because it halts the timer).

closed 2015-08-12 by mlyle