TauLabs/TauLabs #1757: SyncPWM: Jitter Issue at High MPU Sample Rates

tracernz, 2015-07-29:
There seems to be a pretty serious issue with jitter when running SyncPWM at high rates (primarily 1 kHz but also evident at 500 Hz). I've done a little testing with Sparky2, MPU9250 @ 1 kHz sample rate, 184 Hz DLPF. This is the type of thing I'm seeing with output @ 250 us: [image]
This appeared to be happening approximately every 150 ms which is the RFM22b supervisor task rate (I had RFM22b radio disabled but the supervisor task still runs). The raw data I captured is available in [CSV](http://1drv.ms/1KyEUPS) or [MATLAB](http://1drv.ms/1KyExVC) format.

I'm now not so sure that the cause is RFM22b, further investigation required. I adjusted PIOS_RFM22B_SUPERVISOR_TIMEOUT to 230 ms and captured this: [image]
The times on the right side are the time between "glitches" like the one shown. The seem to be quite regular.

peabody124, 2015-07-30:
Nice catch and nice trace. Odd because I did all the initial verification of SyncPWM on sparky2 with that blog post and had scoped it then.

Could you test on master, where Sparky2 is running on FreeRTOS instead of ChiBios? Also, you can disable the PIOS_RFM22B_Init code in pios_board.c so nothing from the modem will be initialized...

tracernz, 2015-07-30:
I just tested the July release, jitter still present :cry: :
[image]

I then checked out master and disabled the RFM22b code in [pios_board.c](https://github.com/TauLabs/TauLabs/blob/next/flight/targets/sparky2/fw/pios_board.c#L868):
```C
#if 0 //defined(PIOS_INCLUDE_RFM22B)
    /* Initalize the RFM22B radio COM device. */
```
No serious jitter, SyncPWM looks to be working well at 1 kHz with the RFM22b code disabled.

Trex4Git, 2015-07-30:
Before i had started development and had a better understanding i also did some test with my Quanton with similar results: https://github.com/bitdump/BLHeli/issues/55#issuecomment-95852167

But since than the OneShot (SnycPWM + HighRes) stuff in Taulabs has improved and i started working on the sub-sampling stuff and added some performance improvements. 
When i looked again with a logic analyzer at the output signal after my changes for the sub-sampling, I think I had not seen such unwanted behavior in the output signal.
But that was after the performance improvements. When I remember correctly before them it looked even worse sometimes when there was high CPU load.

So a made a small PR with just the SPI performance enhancement, and you can test if it has an advantage: https://github.com/TauLabs/TauLabs/pull/1758

peabody124, 2015-07-30:
I can find the spot if you can't, but we should try adjusting the irq priority for the rfm spi irq. That should be preempted by the tim irq

peabody124, 2015-07-30:
Also it would be interesting to add debug code to toggle an ouptut gpio (remove from servo) when entering spi irq

mlyle, 2015-07-30:
It's not the RFM IRQ; he's not using the RFM.  Indeed, we're hitting the supervisor complaining about no interrupts (that code needs to get fixed, and it's what I thought was guilty-- it causes reinit of the RFM, which is expensive, at top task priority).

tracernz, 2015-07-30:
I will try to find time to look into more. I definitely had the RFM disabled so it shouldn't be the IRQ.

peabody124, 2015-07-30:
Oh right gotcha. If things are happening in the supervisor we should use a semaphore to punt then back to the rfm task. That should (probably) not be higher priority that output task (which sets the period).

peabody124, 2015-07-30:
And of course the underlying thing triggering the resets should be fixed. So what mode is the radio in? OLRS or taulink?

mlyle, 2015-07-30:
Right now the RFM task is HIGHEST_PRIO, and there's a rationale for that.

The radio is disabled.

mlyle, 2015-07-30:
But the radio doesn't appear to be the only problem where we're not hitting targets.

PR #1771 (2015-08-05): RFM22b: use PIOS_Thread_Sleep instead of Delay
This avoids a problem where the modem resetting (when disabled)
was tying up in a high priority thread and blocking other system
processes.

Fixes #1757
closed 2015-08-05 by mlyle