TauLabs/TauLabs #1768: GCS Log Warning

jihlein, 2015-08-04:
What does the GCS log warning:

20:46:38[WARNING]file:///C:/TauLabs/build/ground/gcs/share/taulabs/pfd/default/PfdIndicators.qml:17:38: Unable to assign [undefined] to QString

mean?  Seeing this on latest next build, Win7 platform.

[image]

tracernz, 2015-08-04:
The [GCSTelemetryStats.Status property](https://github.com/TauLabs/TauLabs/blob/next/ground/gcs/share/taulabs/pfd/default/PfdIndicators.qml#L17) isn't being copied into QML. It's the same bug as #1764 I think.

tracernz, 2015-08-04:
It's been annoying me too. Would be great if you could test #1769 and confirm it's fixed.

PR #1769 (2015-08-04): GCS: PFD: Fix telemetry status arrow (QT bug 37241)
The telemetry Status property is currently not copied into the QML object properly due to [QT bug 37241](https://bugreports.qt.io/browse/QTBUG-37241). This works around the bug to display the arrow correctly, and avoid spamming the debug gadget with errors. Fixes #1768.
closed 2015-08-04 by mlyle