TauLabs/TauLabs #1770: Flight Mode missing from MAVLink protocol?

OhmEye, 2015-08-05:
Apparently flight mode status is not included in the MAVLink output, so OSDs using MAVLink protocol, specifically minimOSD-Extra are unable to display current flight mode.

mlyle, 2015-08-05:
Note that flight modes as specified by MAVLink are sent, but apparently APM sends a 'custom mode field' that many OSDs respect.

mlyle, 2015-08-14:
The code from mwosd-extra that interprets the custom mode field looks like:

    if (osd_mode == 0) mode_str = "stab"; //Stabilize
    else if (osd_mode == 1) mode_str = "acro"; //Acrobatic
    else if (osd_mode == 2) mode_str = "alth"; //Alt Hold
    else if (osd_mode == 3) mode_str = "auto"; //Auto
    else if (osd_mode == 4) mode_str = "guid"; //Guided
    else if (osd_mode == 5) mode_str = "loit"; //Loiter
    else if (osd_mode == 6) mode_str = "rtl "; //Return to Launch
    else if (osd_mode == 7) mode_str = "circ"; //Circle
    //else if (osd_mode == 8) mode_str = "posi"; //Position Hold (Old)
    else if (osd_mode == 9) mode_str = "land"; //Land
    else if (osd_mode == 10) mode_str = "oflo"; //OF_Loiter
    else if (osd_mode == 11) mode_str = "drif"; //Drift
    else if (osd_mode == 13) mode_str = "sprt"; //Sport
    //else if (osd_mode == 14) mode_str = "flip"; //Flip
    //else if (osd_mode == 15) mode_str = "tune"; //Tune
    else if (osd_mode == 16) mode_str = "phld"; //Position Hold (Earlier called Hybrid)

PR #1795 (2015-08-15): mavlink: send mwosd-extra compatible modes
Attempts to resolve #1770 .  Needs test.
closed 2015-08-15 by mlyle